prusa3d/PrusaSlicer #8177: Possibe fix to corners curling pulling up?

changedsoul, 2022-04-08:
Issues with cooling on corners and having the material cool and shrink, pulling on the lower layers, causing them to detach from the build plate, I had an idea I thought I would bring up and see if its even possible or work.

I was thinking the curling seems to happen for me when its laying down the bottom or top layers. It always starts on outer corners. As it fills in the layer it starts on the outermost edge and works its way in. As it lays it down, its cooling from the outside corners to the inside. I visualize this as whats causing the pulling up on the corners.

I am imagining a simple box for this example.
What if instead, for all solid infills we could start from the center and work our way to the outside. This would possibly put the cooling stress to the center of the part and not the corners thus possibly limiting part warping and edge curling?

Any thoughts?

fe60, 2022-04-12:
@changedsoul
Did you follow the Troubleshooting guide [1] in the Prusa Knowledge Base to avoid warping [2]?

[1] https://help.prusa3d.com/en/article/warping_2011 (Print Quality Troubleshooting - Warping)
[2] https://www.youtube.com/watch?v=HNAtHl6kB7Y (CNC Kitchen - Understand WHY 3D prints WARP)

changedsoul, 2022-04-12:
No not that. 
It seems when it does infills or solid layers, it starts on the corners our outer edges of the model and then works across to the other side. 
i was thinking maybe this pattern was causing the cooling to begin where it starts and maybe contributing to the pulling up of corners. 

If instead of starting on the outer edges, what if it started from the middle and worked it way towards one side, then came back and worked its way to the other side. This would start the cooling in the center and possibly help reduce edges being pulled up?

Sonicboom247, 2022-04-12:
> No not that. It seems when it does infills or solid layers, it starts on the corners our outer edges of the model and then works across to the other side. i was thinking maybe this pattern was [...]

Have you tried using a brim? if not, set it to outer brim only, Brim width anywhere from 6-12 depending on height of model and brim separation to 0

closed 2026-09-29 by prusaslicer-bot[bot]