prusa3d/PrusaSlicer #8179: Toolpath start/finish definition

alexlichius, 2022-04-08:
In PrusaSlicer, I would like to be able to define the start and end point of the toolpath  in each layer. In a closed toolpath both will be identical, but this must not be the case for each layer of the print object.

So far, I use seam position "rear" and rotate the object, e.g. to avoid the toolpath - and stringing - crossing specific interior areas of my print object.

A dedicated function which allows to define toolpath start and end point by clicking at the desired position in each layer would be fantastic

dartrax, 2022-04-09:
> A dedicated function which allows to define toolpath start and end point by clicking at the desired position

Sounds like seam painting, which is already there 😉

lukasmatena, 2022-04-11:
@alexlichius Are you aware of the seam painting option in PrusaSlicer? Is that not enough?

alexlichius, 2022-04-12:
Dear Lukas,
thanks for the hint. Interestingly, the Prusa help chat did not flag this
up to me. I will check it out and let you know if it works for me.
Best regards,
Alex

-- 
Mit freundlichen Grüßen

Alex Lichius

*Diese Nachricht und allfällige Anlagen hiezu sind vertraulich und nur für
die ausschließliche Verwendung durch den/die Adressat/innen bestimmt.* Sollten
Sie nicht die/der angegebene Adressat/in sein, ist das Speichern oder
Kopieren, aber auch jede Offenlegung, Weiterleitung oder sonstige
Verwendung dieser Information nicht gestattet. In diesem Fall verständigen
Sie bitte den Absender und vernichten Sie diese Nachricht.

*This message and any attached files are strictly confidential and intended
solely for use by the addressee(s).* Any storing or copying but also
publication, transmission or other use of the information by a person or
entity other than the intended addressee is prohibited. If you receive this
message by error please contact the sender and delete the material.

lukasmatena, 2024-02-18:
No response, seam painting is already implemented. Closing.

closed 2024-02-18 by lukasmatena