prusa3d/PrusaSlicer #8180: Build plate and UI rendering issue.

peterhieuvu, 2022-04-09:
### Description of the bug

**Problem**
Build plate and plate UI controls are either invisible or rendering strangely in dots.

**Expect**
Build plate and plater UI controls should display normal buttons and complete plate.

**Additional notes:**
Not sure if it's relevant, but I'll drop some of my computer specs in case that matters:
CPU: 5600x
GPU: MSI Ventus 3x 3070

[image]

### Project file & How to reproduce

No project involved. Screenshot above shows an object for reference, but this happens on a clean and empty project as well. Once I open PrusaSlicer, the expected behavior appears for a second before it disappears and I get what I see above.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+win64

### Operating system

Windows 10

### Printer model

MK3S+

dartrax, 2022-04-09:
Looks like #8165 ?

lukasmatena, 2022-04-11:
Closing as a duplicate of #8165.

closed 2022-04-11 by lukasmatena