prusa3d/PrusaSlicer #8183: Filament start GCode not being inserted

tommylux, 2022-04-10:
### Description of the bug

The filament start g-code doesn't appear to be inserted if there are no tool changes.

### Project file & How to reproduce

I have the following G-Code under a filament start g-code:
M165 A98 B1 C1

With a single model selected and the extruder set to #4, the model does not fire off the filament start gcode until like 102019 for some reason, it switches to T3 despite only one extruder selected anyway.

expected outcome: filament start gcode always fires, setting the desired mix ratio.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

241

### Operating system

Windows -10

### Printer model

A20T

rtyr, 2022-04-10:
Please provide your 3mf project file.

tommylux, 2022-04-10:
closing issue: seems to relate to 'multi-colour paint'

closed 2022-04-10 by tommylux