prusa3d/PrusaSlicer #8184: Implement Gosper Curve as a top/bottom infill pattern choice

RampentPotato, 2022-04-10:
**Is your feature request related to a problem? Please describe.**
No, it is a feature request.

**Describe the solution you'd like**
Please implement a top/bottom infill pattern that uses the Gosper space filling curve as described here: 
https://en.wikipedia.org/wiki/Gosper_curve

**Describe how it would work**
Implementing the Gosper Curve as a top/bottom infill pattern would result in an additional aesthetically pleasing infill pattern being made available to the end user.

**Describe alternatives you've considered**
PrusaSlicer already implements the Hilbert Curve space filling curve as an option for top/bottom infill. This is very aesthetically pleasing but creates patterns at 90 degrees, the Gosper curve achieves the same infill result, but uses non 90 degree  pattern.

**Additional context**
None.

jasoncoon, 2024-02-11:
This would be amazing!  Consider this print, which already looks great using the Hilbert curve top fill pattern:

[image][image]

Now imagine it with a more irregular pattern like this:
[image]

jasoncoon, 2026-09-02:
It was a good idea four years ago, it's still a good idea today. 😁