prusa3d/PrusaSlicer #8186: Number of tool changes appears to calculate incorrectly on MMU print

ibuprofane, 2022-04-11:
### Description of the bug

The total number of tool changes appears to calculate incorrectly. I've attached a demo project with 5 painted objects, each 1mm high. At 0.2mm layer height, this means each will have 5 layers. 5 objects x 5 layers should mean 25 tool changes, however PrusaSlicer is reporting 40.

### Project file & How to reproduce

[ToolChangeCountTest.zip](https://github.com/prusa3d/PrusaSlicer/files/8461913/ToolChangeCountTest.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.1+win64

### Operating system

Windows 10

### Printer model

Prusa Mk3s+/MMU2s

lukasmatena, 2022-04-11:
Thanks, you are right. There is a bug introduced more than a year ago, I'm quite surprised nobody reported this earlier (or at least I don't know about it). We'll fix it.

lukasmatena, 2022-04-13:
Should be fixed in [PrusaSlicer 2.4.2-rc2](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.2-rc2). Let us know if it isn't. Closing the issue.

closed 2022-04-13 by lukasmatena