prusa3d/PrusaSlicer #8188: Configuration Wizard "unhandled unkown exception"

Cashew92ftw, 2022-04-11:
### Description of the bug

I wanted to see what outher printers were available as i recently got a new one. So I tried opening the configuration wiard thinking it'll let me see what printers are available and i keep getting an error and the slicer crashes. 

"Unhandled unkown exception: terminating the application"

When I click ok i get  seccond popup

09:15:54: Internal error: Vendor Config Bundle `PrusaSlicer_config_bundle` is not valid: Missing secion or key: `vendor`.
09:15:54: Internal error: Vendor Config Bundle `PrusaSlicer_config_bundle` is not valid: Missing secion or key: `vendor`.

### Project file & How to reproduce

Open prusaslicer, click configuration, and click configuration wizard.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusaslicer-2.4.1+win64

### Operating system

Windows 11

### Printer model

N/A

Cashew92ftw, 2022-04-11:
I havent tried messing around with prusaslicer, but looking at that vendor folder theres 4 files. 

[image]

Perhaps I should uninstall and re-install prusaslicer? I'm still not sure how this came to be which is why ive mentioned it here.

lukasmatena, 2024-02-18:
I guess this is outdated, and because this is the only report of the problem, I assume it was not caused by PrusaSlice,r but by something on your end (corrupted file or whatever). Closing.

closed 2024-02-18 by lukasmatena