prusa3d/PrusaSlicer #8189: MMU Painter Artifacts

BaconFase, 2022-04-11:
### Description of the bug

Colors are appearing where they should not be.

[image]
[image]

### Project file & How to reproduce

[where is the lightgray from.zip](https://github.com/prusa3d/PrusaSlicer/files/8467697/where.is.the.lightgray.from.zip)

Simply slice the file and note the differences between sliced colors and painted colors.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Win10Pro Build19044

### Printer model

MK3S MMU2S

BaconFase, 2022-04-18:
For comparison this is how it looks sliced with 2.4.0
[image]

Versions 2.4.1 and 2.4.2-rc2 both have the strange incorrect color/filament when sliced.

ljford7, 2022-04-22:
I have the same issue. What I believe is happening is that when you "paint" on a color, the Slicer isn't taking the color "down" far enough. Attached are two images. The first one is with the coloring artifacts and the second one showing the lack of color going "down" into the print causing the artifacts (I believe). 

When you paint it on, the color needs to be integrated into the entire piece, just not a superficial surface coloring.

[image]
[image]

BaconFase, 2022-04-22:
@ljford7 I'd need to check out your .3mf but I suspect your issue is simply that you're not painting the sides of the raised lettering.

ljford7, 2022-04-23:
Below is a link to the .3mf file ( I ended up making the back red/white to try and hide some of the artifacts): 

https://drive.protonmail.com/urls/EW6VVS73PM#kyG5HOrFDYo6

BaconFase, 2022-04-23:
@ljford7 There's def something wonky with your stl. 

Even if I change the default color of the model to red; then change the entire backplate to green; then turn off wipe to infill; white still appears in the lettering. 

[image]

Also the triangle shouldn't 'sink' into the model like this.
[image]
.

Seems like there's some infinitesimally small gap between a lot of the lettering sides and the general backplate and also the lettering is segmented into at least 2 parts rather than 1 whole. But I'm not enough of a modeling expert to know how to solve those issues.

ljford7, 2022-04-26:
@BaconFase  I didn't make the .stl so I can't comment on how to fix the issues. I turned a couple of other signs into MMU prints and didn't have the same issues.

lukasmatena, 2022-05-03:
Thanks, we'll look into it.

BaconFase, 2024-02-10:
2.7.2-a2 fixed this issue. Great job yall 👍

closed 2024-02-10 by BaconFase