prusa3d/PrusaSlicer #8190: Flow Extrusion multiplier ignored

Sonicboom247, 2022-04-12:
### Description of the bug

not sure if this is an actual issue or a miscalculation on my part, but when setting the extrusion Multiplier to less than 1 (eg. 0.93) I had over extrusion or just sloppy lines on entire print.  I then started print over with the same file but manually set the flow on the printer using the display down to 93 and achieved a perfect (for my eyes) print.  I thought i would have to raise it back to 100% after the first layer but quality was so good i left it.

This is PLA Plus @ 94% flow, @ 100% flow my extruder was clicking even though gcode has it set to 0.93 which is what sparked my interest.
[image]

### Project file & How to reproduce

[image]

; prusaslicer_config = begin
; avoid_crossing_perimeters = 0
; avoid_crossing_perimeters_max_detour = 0
; bed_custom_model = D:\\Program Files\\Prusa3D\\PrusaSlicer\\resources\\profiles\\Artillery\\bed-x1.stl
; bed_custom_texture = D:\\Program Files\\Prusa3D\\PrusaSlicer\\resources\\profiles\\Artillery\\bed-x1.png
; bed_shape = 0x0,300x0,300x300,0x300
; bed_temperature = 60
; before_layer_gcode = 
; between_objects_gcode = 
; bottom_fill_pattern = monotonic
; bottom_solid_layers = 3
; bottom_solid_min_thickness = 0
; bridge_acceleration = 0
; bridge_angle = 0
; bridge_fan_speed = 100
; bridge_flow_ratio = 1
; bridge_speed = 60
; brim_separation = 0
; brim_type = outer_only
; brim_width = 6
; clip_multipart_objects = 1
; color_change_gcode = M600
; complete_objects = 0
; cooling = 1
; cooling_tube_length = 5
; cooling_tube_retraction = 91.5
; default_acceleration = 0
; default_filament_profile = 
; default_print_profile = 
; deretract_speed = 0
; disable_fan_first_layers = 3
; dont_support_bridges = 0
; draft_shield = disabled
; duplicate_distance = 6
; elefant_foot_compensation = 0.2
; end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
; end_gcode = M77\nG1 E-7 F7000\nM104 S0 T0 ; turn off temperature\nM140 S0 ; turn off bed\nG91  ; relative positioning mode\nG1 Z5;\nG28 X0 Y120 F1200 ; home the X axis and bring the bed to the front\nM18 S5        ; motors off in 5 sec\n
; ensure_vertical_shell_thickness = 0
; external_perimeter_extrusion_width = 0.45
; external_perimeter_speed = 50%
; external_perimeters_first = 0
; extra_loading_move = -2
; extra_perimeters = 1
; extruder_clearance_height = 20
; extruder_clearance_radius = 20
; extruder_colour = #00FFFF
; extruder_offset = 0x0
; extrusion_axis = E
; extrusion_multiplier = 0.93
; extrusion_width = 0.45
; fan_always_on = 0
; fan_below_layer_time = 60
; filament_colour = #FF8040
; filament_cooling_final_speed = 3.4
; filament_cooling_initial_speed = 2.2
; filament_cooling_moves = 4
; filament_cost = 12
; filament_density = 0
; filament_deretract_speed = 25
[... 201 lines cut ...]
; wipe_tower_width = 60
; wipe_tower_x = 180
; wipe_tower_y = 140
; wiping_volumes_extruders = 70,70
; wiping_volumes_matrix = 0
; xy_size_compensation = 0
; z_offset = 0
; prusaslicer_config = end

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Windows 10 X64

### Printer model

Artillery Sidewinder X1 with BTT SKR 2

Sonicboom247, 2022-04-12:
> I don't think your issues are caused by lower extrusion multiplier. Also the difference is only 7%.
> 
> If you have perfect results with multiplier 0.93 + 93% flow on the printer side, then you probably should use multiplier 0.86.
> 
> Anyway, please provide your 3MF project file.

Thanks, here is my project file, any further insight would be greatly appreciated, also i don't understand how you achieved a 0.86 multiplier.  my extrusion calibration came up with a 93.11 for flow, and to be honest i never could wrap my head around how PS does it's calculation for flow but it was my understanding that simply setting the multiplier to .93 is the equivalent to 93% flow.
[Prusa SWX1Gunv2 Grip L test.zip](https://github.com/prusa3d/PrusaSlicer/files/8476157/Prusa.SWX1Gunv2.Grip.L.test.zip)

rtyr, 2022-04-12:
The g-code is OK, extrusion multiplier is not ignored.

Sonicboom247, 2022-04-12:
> The g-code is OK, extrusion multiplier is not ignored.

Thanks for verifying, any idea why i needed to all of a sudden set my flow on the printer to what it is set in PrusaSlicer? also can you explain why you suggest a 0.86 multiplier for a 93.11% flow?

rtyr, 2022-04-12:
You didn't mention 93.11% in previous description and I have no info on how did you get that number. I also don't know what "all of a sudden" means exactly :). I just saw you generated a g-code with 0.93 multiplier and then reduced the flow additionaly on the printer side, therefore it seems you need to use smaller extrusion multiplier or to fix steps/mm in the firmware. You should start by calibrating extruder steps/mm in the firmware and after that calibrate extrusion multiplier for the slicer/material you want to use.

Sonicboom247, 2022-04-12:
Thanks, looks like i mentioned it 3 times in the description, in the screenshot and also in the config posted,"extrusion_multiplier = 0.93".  I actually obtained 93.11 after calibrating my e steps. No worries. i added M221 S93.11 to the start gcode. I just don't understand why i can set the flow rate in Cura to 93.11 and be done.  seems like the flow is always raising questions. https://forum.prusa3d.com/forum/prusaslicer/iso-flow-setting/

rtyr, 2022-04-12:
> like i mentioned it 3 times in the description

You asked why I "suggest a 0.86 multiplier for a 93.11% flow", that was the first time you mentioned that specific value...

It seems that you are changing flow instead of fixing esteps (you can use M92 command for that). Anyway, I dont see any issue related to PrusaSlicer, so I believe  we can close this issue.

closed 2022-04-12 by rtyr