prusa3d/PrusaSlicer #8192: Missing layers on peak shapes

RoberthX8X, 2022-04-13:
### Description of the bug

There are two problems with the slicing of some peaks in small objects:
1)It looks the slicing omits some layers, but it looks it happens the most when the peak is next to other bigger shapes.
2)There is also a bad shape perimeter slicing above those places, that makes very difficult to work with small figures, it doesn't matter if the nozzle diameter used is the smallest one.  If the layers were not omitted I would not complaint about this too.

The problem is that on those places the filament has no good underlaying point for bonding to it, and therefore it leaves a hanging filament which may be straightened and attached to the wiping tower or to some place else. I understand the scaling would help but, the current size is what the customer needs.
[image]
[image]
[32mm_Wendigo_Supported-Soluble.zip](https://github.com/prusa3d/PrusaSlicer/files/8477594/32mm_Wendigo_Supported-Soluble.zip)
I think this is a quality bug, and this fix would improve very much the printer performance.  I would like to see some clean solution to this behavior.  It would be better not to draw those tiny perimeters at all.

Thanks.

### Project file & How to reproduce

[32mm_Wendigo_Supported-Soluble_0.2mm_PLA_MK3SMMU2S_56m.zip](https://github.com/prusa3d/PrusaSlicer/files/8477606/32mm_Wendigo_Supported-Soluble_0.2mm_PLA_MK3SMMU2S_56m.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Windows 11, Windows 10, behavior is the same

### Printer model

i3 MK3S with MMU 2S

rtyr, 2022-04-13:
Those parts are too thin to be printed reliably with your extrusion width. I can see you already enabled "Detect thin walls" option, which can help to print parts, which are too thin, but it can currently cause other issues like this (see the issues with the label "gap fill"). We are working on improvements in this regard.

As for this model, I think at this size and detail, the model is meant to be printed on SLA printer. The result on FDM printer will be probably quite poor even with small nozzle.

dartrax, 2022-04-13:
Seems like a duplicate of #7447

rtyr, 2022-04-13:
There is a lot of similar issues. We will keep it open to test new improvements.

jmerc77, 2022-08-29:
same. thin walls disabled. even so, the thinnest extrusion is smaller than the nozzle. the missing part probably should have shown orange. this is with 2.5.0 alpha3. its obviously too thin. but that's not the main issue. the real problem is its trying and failing to create certain extrusions that should be possible. orange is about .5mm as seen in the screen shot but the gap is missing it.

screenshot of my sliced temp tower:
[image]

DragonFinlay, 2022-09-07:
> same. thin walls disabled. even so, the thinnest extrusion is smaller than the nozzle. the missing part probably should have shown orange. this is with 2.5.0 alpha3. its obviously too thin. but [...]

With 2.5.0 and Arachne, the perimeter generation works entirely different, but I agree that the results look similar.
That being said, your case would be good to have in #8730 which is where we currently keep track of this behavior.

TS232CZ, 2024-05-18:
Same or similar issue with this model https://www.printables.com/model/329778-atomium-brussels-belgium/files
Regardless layer height bottom "half" of the uniformly thick antenna is ignorred by Arachne while the Classic method keeps it perfectly. Only difference in settings between the two is Arachne vs. Classic, all else was the same.
The model part:
[image]
Arachne perimeters:
[image]
Classic perimeters:
[image]

All profiles updated, PS ver. 2.7.4

closed 2026-09-29 by prusaslicer-bot[bot]