prusa3d/PrusaSlicer #8193: SIGSEGV at startup on Manjaro

VincentFrangi, 2022-04-13:
### Description of the bug

I'm using the git aur package of the slicer 
```
Name            : prusa-slicer-git
Version         : 2.5.0.alpha0.r360.gd7c1d0767-1
```

Every time I try to lauch prusa-slicer I got a SIGSEGV:
```
vincent@ElSalvador ~> prusa-slicer
[2022-04-13 13:01:24.358022] [0x00007f8a0bf887c0] [trace]   Initializing StaticPrintConfigs
fish: Job 1, 'prusa-slicer' terminated by signal SIGSEGV (Address boundary error)
```

### Project file & How to reproduce

Just launching the slicer

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.5.0.alpha0.r360.gd7c1d0767-1

### Operating system

Manjaro 21.2.5 Qonos, x86_64 Linux 5.15.32-1-rt39-MANJARO

### Printer model

Prusa-mini

lukasmatena, 2022-04-14:
@VincentFrangi Thanks, but we are not able to reproduce the issue. Our static builds against our dependencies work, including when built against wxWidgets 3.0, so there seems to be no problem in the scenarios we support. We don't have resources to solve issues caused by dynamic linking with undefined versions of all the libraries. The bug does not even have to be in our code.

If you provide a clue about what is wrong or provide a PR that will not break anything else, we can discuss it further. We are not investigating the issue further, sorry.

mapfeld, 2022-07-25:
I encounter the same issue on Manjaro 21.3.5 Ruah and Version 2.4.2-6. Rolling back to Version 2.4.2-4 throws `prusa-slicer: error while loading shared libraries: libwx_baseu-3.0.so.0: cannot open shared object file: No such file or directory`.

lukasmatena, 2022-07-25:
@mapfeld This is caused by the way PrusaSlicer is built in the package, read https://github.com/prusa3d/PrusaSlicer/issues/8299#issuecomment-1186748653 and following posts. It should be solved in 2.4.2-7.

In the future, any bugs appearing in a Linux-packaged PrusaSlicer and not reproducible in our AppImages will be simply closed and left to be solved by the package maintainers. Closing the issue.

closed 2022-07-25 by lukasmatena