prusa3d/PrusaSlicer #8194: Loops no longer connect to brim

dval, 2022-04-13:
### Description of the bug

Loops no longer intersecting with brims. Seems to be present since 2.4.0. Worked in 2.3.3. 

In previous versions I was able to set a loop distance of 2mm and a 3mm brim. This would allow the loop to connect all the pieces. Super useful for printing many small things.  Now, it seems like the supports are counted as part of the object and the loop goes around those as well. This behavior is undesired. 

I would expect the loop to outline the collection of objects, not the additional supports included by slicer.

Previous desired behavior:
[image]

Current undesired behavior:
[image]

### Project file & How to reproduce

Previously I would use the following workflow:

1. Goto Print Settings tab -> click Skirt and brim.
2. In the Skirt section set distance from object to 2mm
3. In the Brim section set Brim width to 3mm

This can't be reproduced in 2.4.x because the 'Distance from object' setting has been replaced with 'Distance from object/brim'.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.0 onward

### Operating system

Windows 10

### Printer model

Any

lukasmatena, 2022-04-13:
You did not provide a project file, but I guess that all you have to do is to set `Print Settings->Skirt and Brim->Draft shield` to Limited. Does it solve the issue?

dval, 2022-04-13:
That does seem to do what I want   Skirt loops settings under draft shield is not intuitive.  But I realize it's my lack of understanding the relationship of the settings.   It seems correct to say the Draft shield is Limited to the Skirt height, drawn with a specified number of loops. It's trivial to set the Draft shield, I just didn't have to before. Thanks.

closed 2022-04-13 by dval

lukasmatena, 2022-04-14:
@dval No, you didn't. But others complained because of various reasons and claimed that the very old behaviour (the one you don't like) was better. There are apparently too many different use cases and the "desirable/undesirable" is relative. That's why we eventually decided to let the user choose.