prusa3d/PrusaSlicer #8195: Problème lié au fonctionnement des modificateurs

Gilles38280, 2022-04-13:
### Description of the bug

[image]

Le modification devrait sur toute la zone du cylindre imposé l'extrudeur 2 (jaune) est bien sur ce modèle sa ne fonctionne pas.

### Project file & How to reproduce

[heart_fond_blanc_modificateur (2).zip](https://github.com/prusa3d/PrusaSlicer/files/8483386/heart_fond_blanc_modificateur.2.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2-rc2

### Operating system

Win 11 Pro

### Printer model

Wanhao D12-300

lukasmatena, 2022-04-14:
This is not an issue. The bottom of the model is actually painted red, except for the smaller part, where it is not painted at all. The modifier is applied before the painting, so it would paint the whole base yellow as you want it to, but the painting then overrides it to red.

Just select the painting tool, use Bucket fill and Shift-click the part on the base that is now red. It will remove the red paint and revert it to default color (also red, so it is difficult to see). The whole bottom will then end up yellow after slicing.

Does it solve the issue?

bubnikv, 2022-04-14:
Ce n'est pas un problème. Le bas du modèle est en fait peint en rouge, à l'exception de la plus petite partie, où il n'est pas peint du tout. Le modificateur est appliqué avant la peinture, il peindrait donc toute la base en jaune comme vous le souhaitez, mais la peinture la remplace ensuite en rouge.

Sélectionnez simplement l'outil de peinture, utilisez Bucket fill et Maj-cliquez sur la partie de la base qui est maintenant rouge. Cela supprimera la peinture rouge et la ramènera à la couleur par défaut (également rouge, donc difficile à voir). Tout le fond finira alors par jaunir après avoir été tranché.

Résout-il le problème ?

Gilles38280, 2022-04-16:
Bonjour, merci de vos réponses. 
Je reconnais manquer de maitrise avec PrusaSlicer et vous remercie pour votre aide. 
Le constat que je fais est que l'ensemble du sujet est certes rouge, j'ai appliqué dans les découpes du texte le blanc pour contraster le texte.  Mon incompréhension est la raison de cette couture de séparation du jaune que j'applique sur quelques millimètres sur l'ensemble de la base du modèle avec le modificateur et cette couture de séparation après le L. 
Aperçu avant le modification
[image]

Gilles38280, 2022-04-16:
Pourquoi à la couche 3 le jaune est-il déjà présent alors que le modificateur avec l'extrudeur 1 est sur une zone de 3 mm ?
Voir captures suivantes :
[image]
[image]

stepikovo, 2022-04-19:
@Gilles38280 can you please provide the current 3MF for us to check it?

Gilles38280, 2022-04-21:
Pas de problème, je vous transmets le 3mf.
Merci

Gilles38280, 2022-04-21:
Pas de problèm
[heart_fond_blanc_modificateur_github.zip](https://github.com/prusa3d/PrusaSlicer/files/8530875/heart_fond_blanc_modificateur_github.zip)
e, je vous transmets le 3mf.

lukasmatena, 2022-04-22:
> Pourquoi à la couche 3 le jaune est-il déjà présent alors que le modificateur avec l'extrudeur 1 est sur une zone de 3 mm ? 
(Why at layer 3 is the yellow already present while the modifier with extruder 1 is on an area of 3 mm?)

As I've already written, the painting has higher priority than the modifier mesh. And because the vertical walls are painted yellow, they are yellow even on the lowest layers, inside the modifier mesh.
(Comme je l'ai déjà écrit, la peinture a une priorité plus élevée que le maillage modificateur. Et parce que les murs verticaux sont peints en jaune, ils sont jaunes même sur les couches les plus basses, à l'intérieur du maillage du modificateur.)

Gilles38280, 2022-04-23:
Bonjour lukasmatena,
Je comprends ton explication sur une priorité plus élevée de la peinture en rapport aux modificateurs. 
Penses-tu qu'il soit possible de modifier cette priorité ou bien est-elle liée à l'ordre d’implantation du modificateur ou de la peinture intérieure qui déterminerait la priorité de l'un ou de l'autre ?

closed 2026-09-29 by prusaslicer-bot[bot]