prusa3d/PrusaSlicer #8197: QUEUE to switch to next "TIP OF THE DAY"

MasterMindBS, 2022-04-14:
Sometimes you have the time to read deeply the TIP OF THE DAY and practice it right now.
I liked it so much - great idea. - the "Filament-Clip"

And if I test and practice this TIP OF THE DAYS. 

I 'd like to have a next TIP OF TODAY or from YESTERDAY, DAY BEFORE YESTERDAY and so on.

It will be very helpful if you can reach a queue of TIPS OF THE DAY (from yesterday of before), because you have now the time and the silence to read and more important to practice these very interesting TIPS right now..

It will be great, if there will be in the TIP OF DAY - boy a bottom "NEXT" or an Sign like an "Arrow" to show the next TIP.

I never learned and practice so much, I didn't know before.
Thanks Dankeschön.

lukasmatena, 2022-04-14:
You can repeatedly use `Help->Show Tip of the Day` to cycle through the tips. Does it fulfill your desires?

MasterMindBS, 2022-10-11:
Dear Lukas,

many thanks for the feedback. 

I didn’t know that additional functionality. Good, but any time, going to HELP - Show TIP oF THE DAY is a little bid too much afford.

Because after Pressing TIP OF THE DAY the HELP Windows appears and any time you are interesting to get more tips you have to read on the bottom, may made some extra information in the TIP-Box and then went back to HELP - TIP OF THE DAY and read the next tip.

Workflow is no easy. But always good that you have the possibility to read more than one TIP OF THE DAY.

Any Arrow-Button in the TIP OF THE DAY-BOX will be better and easier to read more.

Many thanks

Joe Schindler

closed 2026-09-29 by prusaslicer-bot[bot]