prusa3d/PrusaSlicer #8200: Creality Smart Pro Profile Please¡¡¡

Melinix, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

Melinix, 2022-04-15:
My problem is that the creality CR-10 Smart Pro printer profile is still missing...

Create a specific profile for this printer (Please).

lukasmatena, 2022-04-19:
Profiles for 3rd party printers are created and maintained by the community, not by Prusa Research. Anyone can provide a pull request. Closing.

closed 2022-04-19 by lukasmatena