prusa3d/PrusaSlicer #8202: Vase mode: Missing one solid bottom layer

ulab, 2022-04-15:
### Description of the bug

In vase mode you get one less solid bottom layer than configured.

### Project file & How to reproduce

Create a basic shape in a new project, switch to vase mode, slice and count the solid bottom layers in preview.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1+x64

### Operating system

macOS Monterey 12.3.1 (21E258)

### Printer model

Prusa i3 MK3S+ MMU2S

lukasmatena, 2022-04-19:
Thanks, reproduced. The G-Code seems to be fine, the bug is in preview. It shows the first two bottom layers on the first tick of the slider, making an impression that there is one missing.

jessegrunert, 2022-05-06:
I'm not sure this is entirely a preview issue. I'm trying to slice an object in vase mode, layer height is 0.2mm, I have 5 bottom layers configured. The preview is definitely wrong, but the gcode is also wrong only creating infil layers up to 0.6mm layer height. At 0.8mm it starts the spiral up. Expected result should be infill up to 1.0mm.

jessegrunert, 2022-05-09:
[Wine Tumbler (Love for Kelowna Branded) - auto x 5.zip](https://github.com/prusa3d/PrusaSlicer/files/8655031/Wine.Tumbler.Love.for.Kelowna.Branded.-.auto.x.5.zip)

jessegrunert, 2022-05-09:
Bottom layers are set to 5, in the generated gcode, infill layers stop at 0.6mm. I can attach generated gcode if you want as well.

FidelCapo, 2022-08-22:
> Bottom layers are set to 5, in the generated gcode, infill layers stop at 0.6mm. I can attach generated gcode if you want as well.

The number of generated bottom solid layers is limited by the thickness of the bottom of your object. If the bottom thickness was greater, more bottom solid layers will be generated.

FidelCapo, 2022-08-22:
This issue should be fixed.
Closing.

closed 2022-08-22 by FidelCapo

ulab, 2022-08-22:
This is not yet fixed in 2.5.0-rc1.

The preview still shows two layers on 0.2 mm height in vase mode.

FidelCapo, 2022-08-23:
@ulab I'm sorry, my mistake. I forgot to write that the fix will be in the next version. Probably in version 2.6.0.