prusa3d/PrusaSlicer #8204: Inside out Brim

Fra085, 2022-04-15:
When I use brim for adhesion I tend to set it to 5 or 10mm which is quite large. During printing, it starts making it from the desired distance going toward the outside perimeter of the first layer of the model you are printing. This can be good but most of the time, doing concentric perimeters, causes them to be squished in and the last perimeter done of the brim is ugly and fuzzy, transferring these undesirable features to the external perimeter of the model.

It would be great to invert the direction of the brim, building it from the inside out, leaving the best perimeter of the brim touching the external perimeter of the model. This could cause a gap when the first millimetres of extrusion is done but at least you can choose.

Thanks for your wonderful work.

lukasmatena, 2022-04-19:
Isn't this basically an opposite of your issue #8203, where you claim that extrusions after travel moves are bad? If the brim is extruded from the inside, the loop touching the model will be done after a travel. This is the reason why the brim is extruded the way it is - so the outer loops help prime the nozzle.

Are you sure that your issues are not caused by incorrect extrusion rate or first layer calibration? You can also set brim offset in PS 2.4.0 and later.

Fra085, 2022-04-20:
Yes, it is the opposite of issue #8203 because I would be able to use this feature in different conditions and not in combination with skirt. Today I tried adjusting the first layer height and it seems to help. It was not the first time it happened so I thought could be a problem to talk about: it occurred when something was not sticking and I adjusted the settings (bed temperature, first layer height, brim) to make it work. Apparently, I really wanted it to stick to the bed so I ended up squishing it in the bed and adding too much brim. 

Yeah, I know all the features of PS 2.4.0 but offset could not help with better bonding with the brim. I would say this issue can be closed, anyway I would like to experiment with the possibility of changing the brim direction, an expert mode setting, in order to avoid unintentional enabling of it.

Thanks for your help.

closed 2022-04-20 by lukasmatena