prusa3d/PrusaSlicer #8205: Adaptive Retraction Control

Fra085, 2022-04-15:
I don't know if this is the best place to talk about this but I think that if someone could try experimenting with this thing it is the Prusa Team.

Lately, I am printing technical materials, like ASA and CF-PA, and I can't avoid stringing. Yes, materials are new and dried. Ok, for simple models it doesn't show off but for complex ones, with a lot of holes and thin walls, parts I need to print well, it is a curse. I tried every parameter of retraction and filament temperature but nothing worked. 

I started experimenting on what could really help with stringing and the problem seems to be related to the definition of retraction: it does what it tells but it is not enough sometimes. When retracting it removes pressure from the nozzle but while travelling, for a short path too, gravity makes the material inside the nozzle leak out, leaving strings and blobs.

It would be really interesting to test a continuous and adaptive way of retraction during travelling. After all the things done for the retraction (wipe, extra length, retract speed...) I would like to implement a continuous retraction that keeps removing pressure from the nozzle for the whole travel duration. I would need some parameters, like for example:
- continuous retraction speed
- max extra retraction length 
- mm of retraction per mm of travelled distance
Obviously, when printing again after the retraction it is needed to extrude the total retracted length considering the travelled distance. Hope this works.

So the retraction procedure for a 100mm travel could be: wipe, retract 0.8mm, raise Z 0.6mm, travel 100mm at 200mm/s while retracting at 2mm/s and max total retraction length is set to 4mm, lower Z 0.6mm, extrude the total retracted length that is about 3mm considering accelerations.

I am ready for discussion or if this is not a great way to reduce stringing I would like to know what to do because I am not able to avoid them at the moment. If this is not the best place to talk about this just tell me where to try.

Thanks for your help.

bubnikv, 2022-04-19:
What is your printer? Is it a bowden or direct drive?

The issue with retract during move is that depending on your acceleration and jerk settings, one may negatively influence (slow down) the other, so it is tricky to combine them.

Fra085, 2022-04-20:
PrusaSlicer 2.4.1, Prusa MK3s (0.4mm E3D brass nozzle and then brand new 0.4mm E3D hardened steel nozzle), default settings for the machine, 0.15mm QUALITY profile, custom settings for the filament in order to print Carbon Fiber Nylon. I cannot say the exact nozzle temperature or retraction settings because I tried adjusting everything, from default values to extreme ones.

I also tried with materials I always print like PLA and PETG and the problem is the same: yes, the main issue is the model geometry but it is what it is, but I would like to improve the stringing anyway.

This is the CF-PA I am currently experimenting with. I can't show you the printed result because I have already cleaned it but it was like the other example I am showing below.
[image]

This is Prusament Galaxy Black ASA (0.15mm layer height):
[image]

This is common PETG (0.2mm layer height):
[image]

And this is Silk PLA (0.2mm layer height):
[image]

I totally understand these are challenging models but I think this could be an interesting retraction feature, don't know if anybody has ever tried something like what I explained.

Thanks for your help.

Fra085, 2022-04-21:
I am impressed. I will investigate further for HW problems. Can you send me the gcode of that print in order to test if I get the same result?

rtyr, 2022-04-21:
[Twisted_VaseASA.zip](https://github.com/prusa3d/PrusaSlicer/files/8534109/Twisted_VaseASA.zip)

bubnikv, 2022-04-22:
One thing that may help and which may be counterintuitive and found
by @rtyr is that it often helps to reduce Z-hop and retract to speed up
travel to reduce stringing.

Fra085, 2022-04-22:
I run the gcode and it was pretty fine with minor stringing so I am not sure it is a hardware problem. Everything looks fine on the extruder. At the moment I have mounted the hardened nozzle so maybe temperatures could be a little off.

Yes, I tried playing with z-hop and shortening the retraction length too but nothing really worked. Reducing or removing z-hop surely helps with stringing but with a model like this, with lots of holes and thin columns, it ended up destroying the columns.

This is printed with your code with brand new Galaxy Black ASA, just opened yesterday:
[image]
[image]

m-sundman, 2022-04-27:
Retraction speed should be logarithmic, starting with fast retraction and then slower and slower during the move.

closed 2026-09-29 by prusaslicer-bot[bot]