prusa3d/PrusaSlicer #8206: distinguish between "c" filament changes and "p" pauses when reporting "time until" in the display

krhackbarth, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
I pause for the insertion of an RFID tag and then later perform a color change with many of my prints.  The display only shows "hh:mm C".  It's hard to tell if it is reporting the time until the pause or the time until the change.

**Describe the solution you'd like**
use "P" to represent the time until a pause and "C" to represent the time until a change.

**Describe how it would work**
You know the difference between the two and when they will occur.

**Describe alternatives you've considered**
none

**Additional context**
I'm sure you will have to put time in to come up with different values for your supported languages.

rtyr, 2022-05-17:
It would have to be supported by the firmware first.

bubnikv, 2022-05-17:
That means, you need to ask here first:
https://github.com/prusa3d/Prusa-Firmware/issues

closed 2026-09-29 by prusaslicer-bot[bot]