prusa3d/PrusaSlicer #8207: Adding .3mf model and using auto arrange throws model off bed and not visible.

almonster2066, 2022-04-15:
### Description of the bug

When adding a .3mf file to the slicer (2.4.1), and adding multiple instances and then using auto-arrange, it will move the model off the bed and cannot be seen nor found

https://user-images.githubusercontent.com/103776372/163608575-8929b7c9-7862-462f-9d4a-555f9afa5552.mp4

.

### Project file & How to reproduce

[Terminal_X2.zip](https://github.com/prusa3d/PrusaSlicer/files/8497507/Terminal_X2.zip)

1 - start Prusa Slicer 2.4.1
2 - Add the .3mf in the zip file to print using a Prusa MK3S+ default profile
3 - Create 9 instances of the model on the build plate
4 - Hit auto-arrange.  Error message pops up  stating model is not on build plate.
5 - Scroll as far out and you cannot SEE the model anywhere

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.1+win64

### Operating system

Windows 10

### Printer model

MK3S+

foreachthing, 2022-04-15:
Your imported model was a bit "screwy".
I "fixed" it in Blender [untitled-Terminal_X2.zip](https://github.com/prusa3d/PrusaSlicer/files/8498194/untitled-Terminal_X2.zip) and then it worked.

What I've done to it:
- export STL from PS
- placed it at 0 0 5 in Blender
- exported it from Blender with the 3D-Printer addon
- imported it in PS

almonster2066, 2022-04-19:
Regardless of where the location is per the defined position in the .3mf, it is a valid bug since you need to be able to ensure that the function works.  This used to work perfectly fine in prior versions.  I am again reprinting another production part and the same thing  happens.  Please @stepikovo have this fixed.  I believe that lots of people will run into this issue.  I export from F360 and never had this issue with Prusa Slicer before.  I don't use Blender so I cannot speak to that issue.  If you use mainly .stl it seems to be fine but more and more people are using .3mf.

almonster2066, 2022-04-19:
@foreachthing, I did fix the file myself.  The issue is how Prusa Slicer and likely Super Slicer handles .3mf files where x,y and z coordinates are defined.  In the past Prusa Slicer barely handled .3mf and usually put it off the build plate.  This is a fundamental issue where the slicer cannot understand and or remove the defined location and redefine for the defined area of the bed.  Thanks for your post.   The bigger issue is the wider and wider use of .3mf as a standard for solid models.

kubispe1, 2022-11-15:
- 2.5.0 Issue still does not work. 
- 2.6.0a1 should be issue fixed.

kubispe1, 2022-12-14:
I retested again and it is still ok. I am closing this issue with auto-arrange. 
Thank you

closed 2022-12-14 by kubispe1