prusa3d/PrusaSlicer #8208: Totals in Legend/Estimated Print Time

todinskino, 2022-04-15:
I have a suggestion for a simple modification to the Legend/Estimated Print Time window that would be very useful. I suggest adding a line at the bottom of the data series with the totals for each of the columns within the Legend/Estimated Print Time window.

This would be especially useful in determining total material used. It may even negate the need for the total time at the bottom.

Below is an image illustrating how this might look.

[image]

dartrax, 2022-04-17:
Great Idea, I‘ve also missed that in the past.
The totals could even take into account which feature type is disabled. So when you disable visibility of gap fill in the legend, the totals could show print time and so on without gap fill. The percentage not being 100% in this case would be an indication of that.

Vashiru, 2023-01-26:
I think this would be a great addition to PrusaSlicer and given how many tickets have been opened about it over the last few years.

I tend to print with quite a few colors (but often of the same brand(s)) and it's rather easy for me to figure out how much filament is left by putting an empty spool on the scale and then one with a remainder on it to see how much I actually have left. It would help me a lot to have the total weight of filament directly displayed in the slicer so I can figure out exactly how much I I have to scale up/down my model or adjust the settings for the difference between making it and running out before it finishes.

I hope this can be added in 2.6 or 2.7.

Eldenroot, 2023-02-20:
Well, still nothing?!

dartrax, 2023-02-20:
> Well, still nothing?!

You can take a look at Bambu Studio how they implemented it, than make a pull request and implement it in PrusaSlicer!

Eldenroot, 2023-02-20:
I am not paid for it. I just bought Prusa printer and I expect better support for this "premium" printer. Furthermore only few PRs from non core dev team are merged... so waste of time.

drf5n, 2023-08-31:
The totals are shown in the opposite corner of the plater window:

[image]

alexpantone, 2023-09-09:
> The totals are shown in the opposite corner of the plater window:
> [image]

This is true when slicing a model, but not when using the GCode viewer for a gcode file you did not slice yourself.

[image]

dimalei, 2023-11-07:
This seems like a major oversight. The main reason to use a G-Code viewer/analyzer is to see how much material the print eats IMO.

shutosg, 2024-01-21:
I needed this feature too, so I implemented it #12105.
It appears to work fine in my local environment.
I hope it will be merged.

gitcnd, 2024-03-17:
Awesome work @shutosg 
@rtyr ... any chance you can merge that work?

changw-kang, 2024-04-26:
[image]
You can view the weight by selecting the tool item in the Prusa Slicer viewer.

closed 2026-09-29 by prusaslicer-bot[bot]