prusa3d/PrusaSlicer #8209: Not able to preview all Gcode

tommylux, 2022-04-16:
When you use the sliders to preview the Gcode from layer one and at the start of the bottom slider, it is not possible to scroll to the very beginning to preview all start gcode, for instance to review the compiled dynamic gcode. In order to review the start of the gcode, you must export the file and open it up in a separate text editor. 

**Describe the solution you'd like**
Using the bottom slider, to be able to review every line of gcode, or introduce a scroll bar on the Gcode preview.

foreachthing, 2022-04-16:
You have to enable (yes, e-v-e-r-y single time!) "Travel". Then you can go view some more G-Code than usual.
It's ok for some cases - not for every case, thought.
If I really need to check _all_ the g-code, I revert to CraftWare.

tommylux, 2022-04-16:
Would you mind clarifying said ‘Travel’ to enable? Thanks.

tommylux, 2022-04-16:
I found it, thank you.
[image]

foreachthing, 2022-04-17:
I don"t know why one has to enable travel to see more of the G-Code.
And I wish the viewer would remember last settings used.

dartrax, 2022-07-09:
So this can be closed then? For View option defaults #8418 should have this covered.

stepikovo, 2022-07-13:
@dartrax I think those are 2 separate FR, this one is about the gcode preview, #8418 is about remembering the ticked shown options.

dartrax, 2022-07-13:
The question is if ticking "Travel" is an acceptable solution for "to be able to review every line of gcode" like @tommylux requested. If yes, the in my opinion only downside being that it has to be ticked every time (as pointed out by @foreachthing) is covered by #8418. If not, well then this should stay open to be considered in future 😃

closed 2026-09-29 by prusaslicer-bot[bot]