prusa3d/PrusaSlicer #8213: Slicing artifacts introduced by place on face

tg73, 2022-04-18:
### Description of the bug

With a particular model, place-on-face appears to introduce slicing artifacts that are not present in the original STL.

As imported (ie, before place on face):
[image]
[image]

After place-on-face:
[image]
[image]

The "notches" near the corners on the flat face are not expected. The STL is exported from Fusion 360. The fusion model does not have these notches. I've examined the STL in meshmixer, and it looks good and clean to me. The notches also do not show up in PS 3d view - they are visible only when sliced. If I rotate the model in Fusion before exporting, as place on face would do, then PS slices the model without artifacts.

### Project file & How to reproduce

[psissue.zip](https://github.com/prusa3d/PrusaSlicer/files/8504652/psissue.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+win64

### Operating system

Win 10 x64

### Printer model

MK3S+

foreachthing, 2022-04-18:
Question: are your faces perfectly parallel?
When I rearranged it and lowered it just a tiny bit below the bed, it slice just fine.
So, your model has some very small discrepancies.

I tried "simplify model" and decimate to 60% and I got this:
[image]

stepikovo, 2022-04-19:
See [FAQ](https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)), more details in issues #2434 and #1448

closed 2022-04-19 by stepikovo