prusa3d/PrusaSlicer #8214: current filament not saved in project

wbphelps, 2022-04-19:
### Description of the bug

I have several custom filament definitions that I created.  In a project I am working on I select a filement such as "Prusament PC Blend Carbon Fiber @MMU2". I save the project. When I load it again, the filament is now "Prusament PETG @MMU2 - Wm" which is a different filament.  Happens every time.  It appears to be loading the first filament in the print setting list even though the filament I am using for this print is the 3rd one
[prusa save project test 1.zip](https://github.com/prusa3d/PrusaSlicer/files/8509169/prusa.save.project.test.1.zip)
[image]
[image]
.

### Project file & How to reproduce

I tried to export the project following the instructions here. Under File, Export, there is NO project as an option. oops!
I will provide a saved copy of the project file.

Open the project, change the filament, save it. Open it again, filament is not what it was.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+win64

### Operating system

Windows 10 pro

### Printer model

brand new I3 MK3S+ with MMU2S

rtyr, 2022-04-19:
>  It appears to be loading the first filament in the print setting list even though the filament I am using for this print is the 3rd one.

There is no specific extruder selected in your project. The extruder is set to "default", therefore Extruder 1 will be used for that print. I don't see much reason to save "last viewed" filament profile to the 3MF file (that is what is visible on your screenshots).

It looks ok. The arrow shows, where you can assign extruder for the print.
[image]

wbphelps, 2022-04-19:
Thank you,
I selected the extruder in the Print Settings. It isn't clear to me why one place is better than the other, perhaps you can explain?
So what I get from what you wrote is that the "Filament Settings" shown when the project is loaded is not the one that will be used for the print, so that's why this isn't saved in the project.

rtyr, 2022-04-19:
> I selected the extruder in the Print Settings. 

Your print settings were set to extruder 1. But if you mean the selection on your screenshots, it will only show you the filament settings, so you can for example change the temperature etc., it's not meant for assigning the extruder. Note that you can print several objects at the time with different extruders, so it must be specified for the object. 

There is a separate printer profile ".... MMU2S Single", which is meant for single extrusion prints like this. You don't have to set specific extruder with this printer profile, the printer will ask you on the LCD which filament do you want to use for the print.

Feel free to contact our tech-support via live chat at www.prusa3d.com (there is an icon on the bottom-right corner).

Closing.

closed 2022-04-19 by rtyr