prusa3d/PrusaSlicer #8215: Please find a way to allow saving the project name as the name of the long STL file that is not fully displayed

laptop-laptop, 2022-04-19:
**Is your feature request related to a problem? Please describe.**

My STL files have long filename. Even I dragged the vertical line on the right of Name under Infill and Brim on the right of PrusaSlicer, sometimes the entire filename cannot be shown. Is it possible for you to allow one or all of the following:

- By dragging somewhere near the vertical layer bar, expand/contract the visualization area of the printer heatbed on the left and those layer tree with name of the STL file on the right.

or

- During Save As, allow copying of the STL filename and pasting of it when the user tries to save the project filename with the same name of the STL file or by default, propose the Save As filename to be the filname of the STL.

**Describe the solution you'd like**
Please see above.

**Describe how it would work**
Please see above.

**Describe alternatives you've considered**
Please see above.

stepikovo, 2022-04-21:
I'm not sure I understand your request, the name of the model is used in the Project. Can you upload some of your STL files and screenshots of the issue?

laptop-laptop, 2022-07-25:
Sorry I missed this post and created a new one. 

In PS, if the STL filename is too long (for example this_is_an_example_of_long_filename_which_prusa_slicer_cannot_fully_display.stl) , we could drag the vertical bar between Name and Editing to make the app to show more of the filename.

[img]https://i.imgur.com/DWMHRPc.jpg[/img]

However, for files with long filenames, even dragging the vertical bar to the extreme right cannot show the entire name. What I request is to allow us to copy the name of such long STL file (perhaps via right click and copy) and paste the name so that we can save the project file with the same name as the STL file without manually typing the long name.

A new bug that I just discovered is that when I dragged the vertical bar all the way to the right, I cannot drag it back to the left as it disappears from the screen. The label Editing also does not come back. Quitting the app and restarting did not help. I had to load a STL file with short name first for that vertical bar and Editing label to show up again.

[img]https://i.imgur.com/JqOvtnL.jpg[/img]

stepikovo, 2022-07-26:
You can copy the name by double-clicking on the name, however, as I said, the name of the file already is used as the name of the new project automatically.

About the second "issue", you have to use the horizontal scroll bar to get to the rest of the labels, you can then drag them back.

Can we close this issue?

laptop-laptop, 2022-07-26:
Didn't know double-clicking would allow copying. Thank you. 

We can close this issue.

closed 2022-07-26 by laptop-laptop