prusa3d/PrusaSlicer #8216: Vase Mode: Preview of print is not shown

sg-dev1, 2022-04-19:
### Description of the bug

When trying to view the preview of the vase mode print shown in the screenshot, the preview is not shown.
Only some gcode commands on the left are shown. The slider on the bottom to view the tool path is also not shown.

Expected behavior is that the preview is shown and the tool path can also be viewed.

[image]

### Project file & How to reproduce

Opening the project in the described version should show the issue shown on the screenshot.

Exporting the gcode and then reimporting it and viewing with the gcode viewer however works (shows the preview properly).

[vasemode_print.zip](https://github.com/prusa3d/PrusaSlicer/files/8515348/vasemode_print.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1+win64

### Operating system

Windows 10

### Printer model

Custom Config (HyperCube)

neophyl, 2022-04-19:
https://github.com/prusa3d/PrusaSlicer/issues/8032  Might want to read that issue.

sg-dev1, 2022-04-20:
Ok, seems to be the same issue.

I did search for the vase mode tag before, but obviously could not find this issue because it is not tagged as vase mode ...

But thanks anyways for the information.

closed 2022-04-20 by sg-dev1

neophyl, 2022-04-20:
It is tagged as vase mode but as it was closed it will only appear in a search if you dont specify open issues only.  When searching its always a good idea to include closed ones too.