prusa3d/PrusaSlicer #8217: PrusaSlicer 2.4.1 Crashes on opening - Internal Error: invalid STOI Argument

CosmoStarke, 2022-04-19:
### Description of the bug

 have a desktop computer running Microsoft Windows and am getting a popup after opening PrusaSlicer 2.4.1 that states "Internal Error: invalid STOI Argument" when I click okay the application does not progress and closes down.

I only started to get this error after upgrading to PrusaSlicer 2.4.1 and before all was working fine.  I tried uninstalling and reinstalling the application a number of times but got the same error.  Any ideas on what I can do to resolve?

### Project file & How to reproduce

none

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

WIN 10

### Printer model

NA

lukasmatena, 2022-04-20:
Can you please ZIP and post your configuration folder (`C:\Users\user\AppData\Roaming\PrusaSlicer`). It might help us to reproduce the problem. Thanks.

CosmoStarke, 2022-04-25:
Updated with file attachment.  thanks

jmd73, 2022-04-26:
@CosmoStarke What @lukasmatena is asking for is the contents of the configuration folder. Use the path Lukas posted and replace "/user/" with your username on that computer.

It should (at least) contain these folders/files:
[image]

CosmoStarke, 2022-05-01:
K thanks for your patience here is the zip file of that folder
[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/8598978/PrusaSlicer.zip)

bubnikv, 2022-05-04:
@CosmoStarke

Is there anything special about the directory where the PrusaSlicer.ini is being saved? Aren't you using some synchronization service as DropBox, Microsoft OneDrive or Google Drive? Isn't it a network drive? Or maybe you have some back-up service running over that directory?

This PrusaSlicer.ini corruption is something I am completely out of my wits, even with >20 years of development on Windows. It only happens on Windows and only to PrusaSlicer.ini.

hejllukas, 2022-05-04:
@CosmoStarke 

Thank you for providing the ZIP file of the configuration folder.
To get PrusaSlicer working again, you need to delete just the `PrusaSlicer.ini` file located inside the folder `C:\Users\user\AppData\Roaming\PrusaSlicer` (it is the same folder that you was ZIPed and posted in your previous post).
It is safe to delete `PrusaSlice.ini` file, your user profiles will not be affected.
The reason you need to do this is that the `PrusaSlice.ini` file is corrupted, and PrusaSlicer cannot recover from this without deleting `PrusaSlice.ini` file.

We've been facing this issue with corrupted PrusaSlicer.ini for a long time, and we still haven't been able to figure out what's really causing it. We created a workaround to reduce the probability of this issue, and you have been the first one with this issue since then.

Do you remember if one of the following situations happened sometime before the issue occurred:
- The computer was randomly crashed (blue screen of death, for example) or restarted.
- Power outage when the computer was running.
- The computer was manually restarted using the restart button or shut down with a long power button press.

Are you using a notebook or a desktop PC?
- If you're using a notebook, what is its product number (if you know it)?
- If you're using a desktop PC, what type of disk drive does it have? SSD, nVME, or HDD (if you know it)?

CosmoStarke, 2022-05-07:
@ bubnikv
no special synchronization and the files are stored on the C:Drive

@hejllikas 
Deleting the file deleted and all is working again, Thanks for that info.
Q:  Do you remember if one of the following situations happened sometime before the issue occurred?   
A: The computer was manually started using the power button

Q: Are you using a notebook or a desktop PC? 
A: Desktop (alienware X51 R3) with SSD

bubnikv, 2022-06-29:
Very rarely for some unknown reason PrusaSlicer.ini is partially or fully filled in with binary zeros, most likely due to power outage or a bug in hard drive driver. We may only speculate, it may be that Windows kills PrusaSlicer when closing windows, PrusaSlicer does not write the PrusaSlicer.ini file completely or Windows does not flush the disk caches when powering down. We don't know, but we hope we have added another patch to the issue by detecting the null characters in PrusaSlicer.ini as a result of file corruption.

The patch is implemented in PrusaSlicer 2.5.0-alpha2.
Closing.

closed 2022-06-29 by bubnikv