prusa3d/PrusaSlicer #8219: different layer height option just for infill

MrTransparentLeFraude, 2022-04-20:
to improve speed of 3dp the layer height of infill could easily be bigger. make this happen, guys!
thanks!

lukasmatena, 2022-04-20:
Do you mean `Print Settings->Infill->Combine infill every` option? Alessandro Ranellucci already made it happen more than 10 years ago. It was one of the first features that Slic3r got. Does it fit your needs?

MrTransparentLeFraude, 2022-04-20:
yes, this is an awesome option. thanks for pointing it out seeing how i didn't know about this.
however, lemme propose what i had in mind.
given the newer option of variable layer height, this option sets the infill layer height dependent to the whole object layer.
but a fixed thicker infill layer height would be beneficial.
in the realm between 2.5D and 3D this could qualify as about 2.6D

MrTransparentLeFraude, 2022-05-12:
[image]

Jan-Soustruznik, 2025-07-23:
closed as duplicate of #402

closed 2025-07-23 by Jan-Soustruznik