prusa3d/PrusaSlicer #8220: First layer height

tommylux, 2022-04-20:
I would prefer for the first layer that the set height happens after go to start position.

Reasoning:
The hotend travels though a vertical line from the start position at 0.2 height and the heat from the hotend and any oozing makes lines into the magnetic bed.

There isn't really any usable regex to accomplish this myself because it depends if the model has a skirt or a brim and the skirt can be 1 or 2 layers high. Even if there was an option to insert Z position at every G0 G1 command, I'd be able to simply remove the first go to layer (0) to achieve what I want.

stepikovo, 2022-04-20:
Are we talking about the travel move between the purge line and the start of the print? Hotend travels to the start position at Z hop height, so 0.4 mm for Mk3. And the position of that line in the G-code is always the same:
...
G1 E-.8 F2100
**_G1 Z.4 F720_**
;AFTER_LAYER_CHANGE

so it should be replaceable in the gcode.

rtyr, 2022-04-20:
Likely a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/2931.

closed 2022-04-21 by rtyr