prusa3d/PrusaSlicer #8222: ERROR: Some objects are too tall and cannot be printed without extruder collisions.

gearmesh, 2022-04-20:
### Description of the bug

Whenever I select variable layer height, I cannot put more than one object on the bed and print the two or more items sequentially.
Clearly the objects are not too tall and clearly the print nozzle would never hit the other object.

### Project file & How to reproduce

1. Drop an item on the bed
2. Change variable layer
3. Copy and paste the same object on the bed
4. Move objects far away from each other
5. Slice

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.41 and 2.42

### Operating system

Win10

### Printer model

PRUSA i3 MK3S

rtyr, 2022-04-20:
Please provide your 3MF project file.

Likely at least one object is higher than the limit set in print profile (20mm by default).

gearmesh, 2022-04-21:
I'll add it here as a zip
[ERROR.zip](https://github.com/prusa3d/PrusaSlicer/files/8526635/ERROR.zip)
e

gearmesh, 2022-04-21:
This happens with any two or more files I drop on the plate with variable layer height.

rtyr, 2022-04-21:
Your objects are 24.02mm tall, extruder clearance is set to 20mm, so the error message is expected. You have to increase the height clearance if you want to print this. There is no relation to the variable layer height.

See https://github.com/prusa3d/PrusaSlicer/issues/1095 for more info.

Closing.

closed 2022-04-21 by rtyr