prusa3d/PrusaSlicer #8225: Allow auto arrange to put parts inside of parts

rhys100, 2022-04-21:
**Is your feature request related to a problem? Please describe.**
Trying to fit more parts on a bed

**Describe the solution you'd like**
Problem - https://i.imgur.com/bWpadwF.png
Solution - https://i.imgur.com/ik4R3Wk.png

**Describe how it would work**
Auto arranges and fill in all gaps. Could be another option under auto-arrange like in super slicer. https://i.imgur.com/ChB7A41.png

lukasmatena, 2022-04-21:
This is a known issue that was already discussed here, although apparently there is not yet an issue dedicated to track it. Other tasks have priority right now, but hopefully we'll get to it.

> Could be another option under auto-arrange like in super slicer.

Just FYI, the "another option" is also in PrusaSlicer, and it was there first. SuperSlicer pulled it and painted the dialog blue.

5jvm0u4, 2022-07-21:
Same request here, if it's possible, it'll be very useful.

5jvm0u4, 2023-08-24:
Seems to be added into 2.6.2 RC1, very much appreciated, trying it out right now!

lukasmatena, 2023-08-26:
Implemented, closing.

closed 2023-08-26 by lukasmatena