prusa3d/PrusaSlicer #8227: Lightning infill and infill_every_layers don't play nicely together.

plampix, 2022-04-21:
### Description of the bug

Slicing a calibration cube with lightning infill and infill_every_layers set to 1 works great, but when you increase it, the infill will be quite random.

### Project file & How to reproduce

[combine every X layers combined with lightning infill.zip](https://github.com/prusa3d/PrusaSlicer/files/8530727/combine.every.X.layers.combined.with.lightning.infill.zip)

Left cube is normal, the right one has combine_every_layers set to 2
[image]

I've hidden perimeters in the preview, so you can actually see what I mean.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.5.0-alpha0

### Operating system

Debian GNU Linux 11

### Printer model

Prusa MK3S+, Ender 3, Custom printer.

lukasmatena, 2022-04-21:
@plampix Thanks, reproduced. We will look into it.

hejllukas, 2022-05-05:
Thank you for reporting this issue. It should be fixed with 09a9d79e993baa55975e853affd25c22414c45f2.

closed 2022-05-05 by hejllukas