prusa3d/PrusaSlicer #8228: Model errors icon does not go away after proper model reloading

PavLeMa, 2022-04-21:
### Description of the bug

Open model with errors. You'll get an exclamation mark in triangle next to model name and message like "200 open edges" under Info. Correct model file with other software, save model file without errors with same name. Right-click model name, select "Reload from disk". Watch exclamation mark in triangle next to model name and "No errors detected" under Info.
[image]
.

### Project file & How to reproduce

Open model with errors. You'll get an exclamation mark in triangle next to model name and message like "200 open edges" under Info. Correct model file with other software, save model file without errors with same name. Right-click model name, select "Reload from disk". Watch exclamation mark in triangle next to model name and "No errors detected" under Info.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.4.1+win64-202203101054

### Operating system

Windows 7 (build 7601, Service Pack 1), 64-bit edition

### Printer model

Anet

FidelCapo, 2022-04-22:
Thank you for the report. I am able to reproduce this issue. The exclamation mark icon disappears if the object is deleted and loaded again.

FidelCapo, 2022-05-03:
This issue should be fixed in future versions.
Closing.

closed 2022-05-03 by FidelCapo