prusa3d/PrusaSlicer #8230: External perimeters first do not work

RobertoRock, 2022-04-22:
### Description of the bug

It always prints the outer perimeter, whether or not the box is checke
[image]
[my system.txt](https://github.com/prusa3d/PrusaSlicer/files/8535739/my.system.txt)
[image]
[image]
d

### Project file & How to reproduce

[base.zip](https://github.com/prusa3d/PrusaSlicer/files/8535765/base.zip)
first slice, second see the first layer, and see with the nozzle position tool how it always prints the outer perimeter first

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer 2.4.1

### Operating system

Windows 11,  os X el capitan same problem

### Printer model

Creality Cr10 V3 nozzle 4mm

The-Old-Tom, 2022-04-27:
Is this only on the first layer? 
I assume that this desired behaviour as you want to print next to the brim on the first layer to improve adhesion, though the designers will have to confirm this.
Subsequent layers **should** print according to the 'External Perimeter First' setting, though I have found a bug that occurs where perimeters interfere with each other https://github.com/prusa3d/PrusaSlicer/issues/8154. This doesn't seem to be the case here

dartrax, 2022-05-01:
He has „Detect Thin Walls“ enabled, so this could also be related to https://github.com/prusa3d/PrusaSlicer/issues/7209, https://github.com/prusa3d/PrusaSlicer/issues/6909.
I‘ve heard that developers want to imrpve that in 2.5

closed 2026-09-29 by prusaslicer-bot[bot]