prusa3d/PrusaSlicer #8231: Prusa Slicer v2.4.1 missed large chunk of object

GRIFFCOMMca, 2022-04-22:
### Description of the bug

Import of a complete STL file, shows correctly in preview, when generating the code appears to just miss a large section of the print.  RThe STIL file loads ok in online viewer, inside Pruce Slicer and Sketchup Pro 2022, so no idea what's going on.  The original CAD file sides are perfect.

### Project file & How to reproduce

[](url)Import the STL file then preview before printing, large section has not been sliced.  The amount missing depends which way on the bed its sitting:

[STL file download](https://www.ashleygriffin.ca/temp/3dprint/2022-04-21-brokenprint.stl)

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Windows v2.4.1 x64

### Operating system

Windows 10 Pro 21H2

### Printer model

Prusa MK3s with MMU2s

neophyl, 2022-04-22:
PS does say there are 1106 open edges after loading it and shows the warning triangle. That is after auto repairing 1164 errors.   So its obvious that the model is broken.  
Using the right click and fix through netfabb option repairs it, allowing a successful slice.  Also fixing it in Blender worked for me too.

GRIFFCOMMca, 2022-04-22:
Appears this ticket can be closed.

closed 2022-04-22 by GRIFFCOMMca