prusa3d/PrusaSlicer #8232: Z hop axis layer changes inside seam alignment drags filament across inside diameter of thin walls, though outside thin wall perimeters print without dragging filament.

GI101, 2022-04-22:
### Description of the bug

Cura, Prusha, Slick3R, print the ID of very thin wall (.038") by dragging filament across top of the inside wall for several centimeters before PETG filament finally sticks roughly at layer >25+. Issue occurs with or without retraction or wipe being enabled in Z axis seam changes only. It seems extruders Z stop correction (-0.275mm) is not accounted during layer changes of any Z seam type chosen but only effecting inside surface of thin walls. Using RepRap Repetier G code flavor and auto filament cooling enabled for TronXY5SA printer.

### Project file & How to reproduce

There is no export project config selection in the Prusha, File Export dialog box. The Prusha control bundle.ini and SD file Gcode 
[SealCollarSD_PETG_Prusha.zip](https://github.com/prusa3d/PrusaSlicer/files/8542763/SealCollarSD_PETG_Prusha.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusha 2.3.1 Win64

### Operating system

Windows 7

### Printer model

TronXY5SA 330mm x 330mm x 400mm Glassplate with black auto level sensor

rtyr, 2022-04-22:
Please provide your 3MF project file (File - Save project as...).

> Cura, Prusha, Slic3R

I don't understand why are you mentioning Cura?

GI101, 2022-04-22:
There is no Prusha project file needed for Repetier. We only use Prusha/Cura/Slic3R slicers to generate Gcode for Repetier SD file output. All slicers included with Repetier have the same thin inside wall issue on Z axis seam alignment changes. It almost appears the extruder goes in straight line during z layer seam changes but on the inside wall only and eventually follows the diameter curve of the inside wall.

rtyr, 2022-04-22:
> There is no Prusa project file needed

It is needed, that is why we are asking for it. Frankly I am not sure I understand your description and why do you think it is slicer related issue, but I can check the output.

GI101, 2022-04-22:
Added the 3MF project ZIP above. Oddly inside wall dragging filament is not an issue when the wall is only 1 perimeter thick. It doesn't matter how the Z seam position is made to show it drags lines immediately after Z axis changes walls 2 parameters with center filled. Yet printer makes perfect circles <0.0005" difference X/Y diameter via digital calipers. I first thought Z hop (0.25mm) on/or retraction on hop causing issues and reduced hop to (0.125mm) even disabled retraction hop made no difference.

rtyr, 2022-04-22:
I think it is probably just misconfigured profile vs printer FW. The g-code seems ok (except few top layers with non-ideal gap fill).

I can see you are using z-offset setting, which may generate Z heights, which are below the limit set in your firmware (first layer is printed at Z0.045 for example). You should rather solve z-offset on the printer side. 

Other potentionally related settings:
- "only retract when crossing perimeters" option is enabled, therefore retractions (and z-hops) are not performed inside the walls
- there are some retraction overrides set in the filament profile so retraction settings in the filament profile are always used, no matter what is set in the printer profile. So for example if you completely disabled the retraction in the printer profile, it had zero effect. 

> I first thought Z hop (0.25mm) 

Z-hop is set to 0.15mm in your project file (0.25mm is set in the printer profile, but it is overriden by filament profile)

GI101, 2022-04-24:
Even odder is extruder travel indicator line for the thin wall is shown to be exiting above each layer Z seam change. The -Z offset was set to compensate for updated printer opaque level sensor and thicker lattice glass replacing thinner removable aluminum plate with plastic stick-on build sheet. The Zero offset is not being saved, after a home position command it seems to forget the firmware zero offset parameter and drags lines on the build plate.

This printer had the same thin wall Z seam issues even with the original level sensor, aluminum build plate and sticky sheet it would drop lines randomly along Z seam alignment changes. That produced random voids in the perimeter walls more noticeable in thinner walls than in thicker ones.

rtyr, 2022-04-24:
Probe offset should be set on the firmware level. It can be set by M851 command. Z-offset setting in PS only reduces Z values in the gcode output, it is a workaround sort of.

Your issues may be related to excessive Z travel speed (skipped steps), you may try to reduce it in print settings.

GI101, 2022-04-25:
So without printer defaults set as before and Z hop 0.125mm only on retraction made little difference with 80mm/sec travel speed. Reduced online 80mm travel speed down to 24mm/sec, Z seam line drops were slightly less but still drags lines across the diameter .75" down to .5"at slower speed. Oddly at times the perimeter speed ended up being 16mm/sec going into Z seam thin wall print and accelerating to 24mm/sec inside Z seam travel and often 24mm/sec coming out of the aligned  Z seam.  I have springs on the core XY belts near extruder hot end to keep backlash to minimal on motor reversals.

Other observation being the center fill between two perimeter walls did not appear to print in the upper thin wall regions. There is some Z perimeter wall undercutting in horizontal lines being printed then suddenly XY returns to print outside perimeter above the lower region lines that were just undercutting. I reset the printer to defaults and recalibrated the auto leveling and set the Z offset fairly tight below A4. The prior print job brim lines did not iron/fuse together even with -0.275mm offset in printer settings, brim was not connected to lower 1st layer. It may be the elephant print setting is over riding the Z -0.275mm offset height in the 1st layer. That did not occur via Slic3R print as slicing algorithm did not have elephant foot setting.

GI101, 2022-04-25:
The updated 3mf project prints ABS filament:
[PolyNitrileSeal_Collar.zip](https://github.com/prusa3d/PrusaSlicer/files/8558019/PolyNitrileSeal_Collar.zip)

rtyr, 2023-09-12:
I am closing this as it does not seem slicer related.

closed 2023-09-12 by rtyr