prusa3d/PrusaSlicer #8233: Bug: Error Message Flow::with_spacing() with small nozzle

rocket-plane, 2022-04-22:
### Description of the bug

Problem: Wenn trying to slice the selected file the error message:
[image]
is displayed.
[image]

Expected: Sliced without Problems

### Project file & How to reproduce

[Project & Config.zip](https://github.com/prusa3d/PrusaSlicer/files/8543610/Project.Config.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.4.1 + wind64

### Operating system

Windows 7 x64

### Printer model

Prusa i3 MK3S

rocket-plane, 2022-04-22:
OK, I found out the background. The model had an adaptive layer height. And some layers where to thick for the 0.25mm nozzle. Changing back to constant layer hight or a new click on "adaptive" in the layer hight control helped.

It would be nice to get a meaningful error message. e.g. "Some layers are to thick. Please also check adaptive layer settings" 
When changing the settings, it could be checked that the layers are not thicker than the configured maximum value. If not show a warning.

faith-healer, 2022-05-03:
if I set some options in "Print Settings - Speed - Speed for print moves" in % I get same error, but if I set them in absolute numbers it fixes the issue.
either way it's not evident behavior, so I think it's a bug.

closed 2026-09-29 by prusaslicer-bot[bot]