prusa3d/PrusaSlicer #8234: Support points in the middle of the air (SLA)

robingantois, 2022-04-23:
### Description of the bug

release 2.4.1
After automatically adding support (SLA) and when I want to add support points manually afterwards, support points are only placed in the middle of the sky. Not at all at the indicated place on the model!
[image]
[image]

### Project file & How to reproduce

[Main body.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/8546463/Main.body.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1 +X64

### Operating system

Windows 10 & IOS Monterey

### Printer model

SL1

lukasmatena, 2022-04-23:
Looks like #6100. Can you please try in 2.4.2-rc2? Thanks.

robingantois, 2022-04-23:
**Update: I haven't tried 2.4.2-rc2 yet! But what does work is the following:

Load the model, orient it, export it as stl. Close Prusaslicer. Reopen the exported model again and then the problem does not occur!

lukasmatena, 2023-02-06:
This is almost certainly a duplicate of #6100, which is long fixed. Closing.

closed 2023-02-06 by lukasmatena