prusa3d/PrusaSlicer #8236: Strange relation between support, extruder1, extruder2, skirt and the variable current_extruder

JoeScotty, 2022-04-23:
### Description of the bug

Have strange things happening in Prusaslicer, which I didn't watch in previous versions.
Using Prusaslicer for years, configured for my self  made cartesian printer (Dual Extruder, BLTouch, Filament sensor, Rumba board)

If no support is set, everything works fine:
- select Extruder1, no skirt is created, and it prints using extruder1
- select Extruder2, no skirt is created, and it prints using extruder2

If I select support everywhere, strange things happen:
- select Extruder1, no skirt is created, and it prints using extruder1
- select Extruder**2**, **skirt** is created, and it prints using extruder**1**

### Project file & How to reproduce

Use the attached project, and just switch from Extruder1 to extruder 2, and
check support or uncheck support

I have user defined start GCode, which uses different gcodes depending on selected extruder.
This worked for years now, so should be fine. The last part is shown here:

        ; retract unused extruder 1mm
        {if current_extruder == 0}
        M109 S0 T1  ; unused extruder stop heating
        T1
        {endif}
        {if current_extruder == 1}
        M109 S0 T0  ; unused extruder stop heating
        T0
        {endif}
        G92 E0 ; Extruder koordinate auf 0
        G1 E-0.2 F3000 ; inaktiver Extruder 0.2mm retract
        T[current_extruder]
        ; User defined Start G-Code End ----

The variable current_extruder seems not to be set correctly, if extruder 2 and support is activated.
As well skirt (or is it the wind shield?) is activated, if extruder 2 and support is activated

Project file:
[Bug Box.zip](https://github.com/prusa3d/PrusaSlicer/files/8547615/Bug.Box.zip)
[image]
[image]
[image]

 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

Win 10 21H2

### Printer model

self made cartesian

JoeScotty, 2022-04-24:
Some more information:
I wrote that in case of selected extruder2 (E2) the print is made using extruder1.
After checking the gcode file in detail I found, that E2 is used for part printing, and E1 is used for support and partially for skirt.

And there is a workaround for printing completely with E2 including support and without skirt:
- select E2 in print bed view
- also change the print settings - several extruders, and change all five Extruder settings to 2
in this case no unwanted skirt is created and only E2 is used for printing.

My extruder settings (did not change since several prusaslicer versions):

[image]

In summary the strange relation (see issue topic) can be reduced to:
- In previous versions, when switching from E1 to E2, I just had to switch in print bed view and change the extruder.
- In the current version, when switching from E1 to E2, I additionally have to change all 5 extruder settings in print settings/several extruders

Still strange: a skirt in full height is created, if I switch from E1 to E2 only in print bed view.

rtyr, 2022-04-25:
Set both these values to 0, it should solve your issues.

[image]

JoeScotty, 2022-04-25:
That solved the issue!

Thanks a lot and have a nice week!

closed 2022-04-25 by JoeScotty