prusa3d/PrusaSlicer #8237: Print all first-layers of sequential objects at the same time

MechanicalCat, 2022-04-24:
When sequentially printing PET-G objects ("Complete Individual Objects"), the second object almost always detaches from the bed because its first layer is printed at the wrong (too high) temperature. (For PET-G, first-layer temp should be 85C/240C, other layers are 90C/250C).

How about giving us an option to print ALL THE FIRST LAYERS OFF ALL THE OBJECTS right at the start of the print-run? Thereafter to print the objects sequentially on top of those well-bonded first layers.

Bug #877 is related to this problem. The bug's resolution was to use "between objects custom G-code" to add M190/M109 commands so the printer waits for the temperatures to change between objects. This isn't a good solution: we get oozing which itself causes the second sequential object to fail.

rtyr, 2022-04-25:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3095, I am closing this one.

closed 2022-04-25 by rtyr

MechanicalCat, 2022-05-10:
To test just how critical this issue is for PET-G sequential printing, I have now written a (terrible!) bash script which re-orders the PET-G gcode output from PrusaSlicer. The script simply prints all the first layers of all sequential objects non-sequentially at the start, then it raises the temperatures and prints all the objects (layers 2+) sequentially. This means that the first layers are reliably printed at their proper (lower) temperatures without needing a cooldown pause (and consequent oozing) between them.
Normally - with PrusaSlicer's raw sequential output - my second sequential object would almost invariably fail to stay bonded. But, using this script, I can now print 4-object sequential PET-G prints completely successfully.
So, including this small mod would result in significant reliability improvements to Prusa printers, when printing PET-G sequentially.

secv, 2022-10-09:
Use Brim in current version. The Brim will print first before any objects. https://github.com/prusa3d/PrusaSlicer/issues/3095#issuecomment-1272611737

JimmyFLASH-CZE, 2023-01-16:
Voting for the idea.
1st layer to the most vulnerable for failure.
Please add this feature.

foreachthing, 2023-07-31:
Got an post processing script working (I hope!) over here: https://github.com/prusa3d/PrusaSlicer/issues/11043