prusa3d/PrusaSlicer #8238: Boost 1.79 deprecated boost::filesystem::ofstream

bnavigator, 2022-04-24:
### Description of the bug

Compiling PrusaSlicer 2.4.1 with boost 1.79 results in

```
[   22s] [ 13%] Building CXX object src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o
[   22s] cd /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/build/src/hints && /usr/bin/g++-11 -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOST_LOG_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src -I/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/build/src/platform -isystem /usr/include/eigen3 -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -Werror=return-type -flto=auto -g -DNDEBUG -fext-numeric-literals -Wall -Wno-reorder -O2 -g -DNDEBUG -fPIE -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -std=gnu++17 -MD -MT src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o -MF CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o.d -o CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o -c /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp
[   22s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp: In function 'bool write_to_pot(boost::filesystem::path, const std::vector<std::pair<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> > >&)':
[   22s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp:12:28: error: 'ofstream' is not a member of 'boost::filesystem'
[   22s]    12 |         boost::filesystem::ofstream file(std::move(path), std::ios_base::app);
[   22s]       |                            ^~~~~~~~
[   22s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp:12:28: note: suggested alternatives:
[   22s] In file included from /usr/include/c++/11/ios:38,
[   22s]                  from /usr/include/c++/11/ostream:38,
[   22s]                  from /usr/include/c++/11/iostream:39,
[   22s]                  from /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp:1:
[   22s] /usr/include/c++/11/iosfwd:165:41: note:   'std::ofstream'
[   22s]   165 |   typedef basic_ofstream<char>          ofstream;
[   22s]       |                                         ^~~~~~~~
[   22s] /usr/include/c++/11/iosfwd:165:41: note:   'std::ofstream'
[   22s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp:20:17: error: 'file' was not declared in this scope
[   22s]    20 |                 file << "\n#: resources/data/hints.ini: ["<< element.first << "]\nmsgid \"" << element.second << "\"\nmsgstr \"\"\n";
[   22s]       |                 ^~~~
[   22s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp:22:9: error: 'file' was not declared in this scope
[   22s]    22 |         file.close();
[   22s]       |         ^~~~
[   22s] make[2]: *** [src/hints/CMakeFiles/hintsToPot.dir/build.make:79: src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o] Error 1
[   22s] make[2]: Leaving directory '/home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/build'
[   22s] make[1]: *** [CMakeFiles/Makefile2:1109: src/hints/CMakeFiles/hintsToPot.dir/all] Error 2
```

The release notes of boost::filesystem say:

https://www.boost.org/users/history/version_1_79_0.html
> Deprecated: boost/filesystem/string_file.hpp header is deprecated and will be removed in a future release. The header is no longer included by boost/filesystem.hpp by default. Users are advised to implement the functionality themselves or migrate to other implementations.

A workaround is to add the include explicitly, but you might want to consider replacing the boost function by the equivalent from the standard lib (plus parameter type handling):

```diff
Index: PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp
===================================================================
--- PrusaSlicer-version_2.4.1.orig/src/hints/HintsToPot.cpp
+++ PrusaSlicer-version_2.4.1/src/hints/HintsToPot.cpp
@@ -2,6 +2,7 @@
 #include <vector>
 #include <string>
 #include <boost/filesystem.hpp>
+#include <boost/filesystem/string_file.hpp>
 #include <boost/dll.hpp>
 #include <boost/property_tree/ini_parser.hpp>
 #include <boost/nowide/fstream.hpp>
```

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.1 source

### Operating system

openSUSE Tumbleweed

### Printer model

N/A

bubnikv, 2022-04-25:
Please let me know whether #8238 helps.

The original code tried to avoid conversions from / to UTF8 on Windows.

I have replaced boost::filesystem::fstream with boost::nowide::fstream
variants with the unfortunate cost of string path conversion on Windows
from 16 bits to UTF8 and back to 16 bits.

Unfortunately we cannot use std::filesystem yet as it is missing
on older MACs and because the interface is crooked minefield on Windows
see https://github.com/microsoft/STL/issues/909

bnavigator, 2022-04-25:
408e56f does not include the fix for the fff_print test:

```
[  221s] [ 60%] Building CXX object tests/fff_print/CMakeFiles/fff_print_tests.dir/test_data.cpp.o
[  221s] cd /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/build/tests/fff_print && /usr/bin/g++-11 -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME>
[  221s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/tests/fff_print/test_data.cpp: In function 'std::string Slic3r::Test::gcode(Slic3r::Print&)':
[  221s] /home/abuild/rpmbuild/BUILD/PrusaSlicer-version_2.4.1/tests/fff_print/test_data.cpp:289:21: error: variable 'std::ifstream t' has initializer but incomplete type
[  221s]   289 |     std::ifstream t(temp.string());
[  221s]       |                     ^~~~
[  221s] make[2]: *** [tests/fff_print/CMakeFiles/fff_print_tests.dir/build.make:93: tests/fff_print/CMakeFiles/fff_print_tests.dir/test_data.cpp.o] Error 1
```

bubnikv, 2022-04-25:
Will this help? 926ae0471800abd1e5335e251a5934570eb8f6ff

bnavigator, 2022-04-25:
Yes the additional commit helps. I cannot speak for Windows or MAC builds, but on openSUSE Linux, PrusaSlicer now compiles with boost 1.79. Thank you.

I am no C++ expert, but I guess since `boost::filesystem` stopped to include the `fstream` headers automatically, there was no type information for `std::ifstream` available anymore.

closed 2022-04-25 by bnavigator

bubnikv, 2022-04-25:
thanks for help.

mkyral, 2022-08-15:
I just run into this issue in 2.5.0 beta1. 

```
[32/364] /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOST_LOG_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1_build/src/platform -isystem /usr/include/eigen3  -DNDEBUG -march=native -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder -fPIE -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -std=gnu++17 -MD -MT src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o -MF src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o.d -o src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o -c /var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp
FAILED: src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOST_LOG_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DSLIC3R_GUI -DUNICODE -DWXINTL_NO_GETTEXT_MACRO -D_UNICODE -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src -I/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1_build/src/platform -isystem /usr/include/eigen3  -DNDEBUG -march=native -O2 -pipe -fext-numeric-literals -Wall -Wno-reorder -fPIE -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -std=gnu++17 -MD -MT src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o -MF src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o.d -o src/hints/CMakeFiles/hintsToPot.dir/HintsToPot.cpp.o -c /var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp
/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp: In function ‘bool write_to_pot(boost::filesystem::path, const std::vector<std::pair<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> > >&)’:
/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp:12:28: error: ‘ofstream’ is not a member of ‘boost::filesystem’
   12 |         boost::filesystem::ofstream file(std::move(path), std::ios_base::app);
      |                            ^~~~~~~~
/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp:12:28: note: suggested alternatives:
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/include/g++-v11/ios:38,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/include/g++-v11/ostream:38,
                 from /usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/include/g++-v11/iostream:39,
                 from /var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp:1:
/usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/include/g++-v11/iosfwd:165:41: note:   ‘std::ofstream’
  165 |   typedef basic_ofstream<char>          ofstream;
      |                                         ^~~~~~~~
/usr/lib/gcc/x86_64-pc-linux-gnu/11.3.0/include/g++-v11/iosfwd:165:41: note:   ‘std::ofstream’
/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp:20:17: error: ‘file’ was not declared in this scope
   20 |                 file << "\n#: resources/data/hints.ini: ["<< element.first << "]\nmsgid \"" << element.second << "\"\nmsgstr \"\"\n";
      |                 ^~~~
/var/tmp/portage/media-gfx/prusaslicer-2.5.0_beta1/work/PrusaSlicer-version_2.5.0-beta1/src/hints/HintsToPot.cpp:22:9: error: ‘file’ was not declared in this scope
   22 |         file.close();
      |         ^~~~
```

stepikovo, 2022-08-24:
@mkyral Is it the same for 2.5.0-rc1?

bnavigator, 2022-08-24:
The above mentioned commits which fix it are only in the master branch and 2.6.0-alpha, not in 2.5.0

mkyral, 2022-08-25:
@stepikovo: yes, rc1 also needs boost 1.79 patches to build. As I have found, it is intended:
https://github.com/prusa3d/PrusaSlicer/issues/8451#issuecomment-1181858700

eduncan911, 2024-09-20:
I believe we have a regression here @bubnikv 

On 2.8.1-rc1, docker ubuntu 24.04, I am getting these errors again:

```
/kasai/PrusaSlicer-version_2.8.1-rc1/src/slic3r/GUI/Gizmos/GLGizmoSVG.cpp: In member function ‘void Slic3r::GUI::GLGizmoSVG::draw_filename()’:
/kasai/PrusaSlicer-version_2.8.1-rc1/src/slic3r/GUI/Gizmos/GLGizmoSVG.cpp:1598:32: error: ‘ofstream’ is not a member of ‘boost::nowide’; did you mean ‘std::ofstream’?
 1598 |                 boost::nowide::ofstream stream(out_path_str);
      |                                ^~~~~~~~
In file included from /usr/include/c++/13/bits/std_thread.h:38,
                 from /usr/include/c++/13/thread:45,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/deps/build/destdir/usr/local/include/oneapi/tbb/detail/_machine.h:51,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/deps/build/destdir/usr/local/include/oneapi/tbb/detail/_utils.h:26,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/deps/build/destdir/usr/local/include/oneapi/tbb/scalable_allocator.h:22,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/src/libslic3r/../libslic3r/Point.hpp:15,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/src/slic3r/GUI/Gizmos/GLGizmoBase.hpp:8,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/src/slic3r/GUI/Gizmos/GLGizmoSVG.hpp:6,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/src/slic3r/GUI/Gizmos/GLGizmoSVG.cpp:1:
/usr/include/c++/13/iosfwd:167:41: note: ‘std::ofstream’ declared here
  167 |   typedef basic_ofstream<char>          ofstream;
      |                                         ^~~~~~~~
/kasai/PrusaSlicer-version_2.8.1-rc1/src/slic3r/GUI/Gizmos/GLGizmoSVG.cpp:1599:21: error: ‘stream’ was not declared in this scope; did you mean ‘std::io_errc::stream’?
 1599 |                 if (stream.is_open()){
      |                     ^~~~~~
      |                     std::io_errc::stream
In file included from /usr/include/c++/13/ios:44,
                 from /usr/include/c++/13/istream:40,
                 from /usr/include/c++/13/sstream:40,
                 from /kasai/PrusaSlicer-version_2.8.1-rc1/src/libslic3r/../libslic3r/Point.hpp:21:
```

lukasmatena, 2024-09-20:
Those are not the same error. It is not related and it is caused by missing includes. Fixed in 2.8.1.