prusa3d/PrusaSlicer #8239: Gap fill not working when using Auto Speed

ahmetcemturan, 2022-04-25:
### Description of the bug

Gap fill does not work when speed is set to zero as needed for the AutoSpeed to work.
When just speed for Gap Fill is entered Auto Speed is not working (Prints with ultra slow settings)

### Project file & How to reproduce

[yuvarlak kap exp.zip](https://github.com/prusa3d/PrusaSlicer/files/8552006/yuvarlak.kap.exp.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.2 rc2  same on 2.4.1

### Operating system

Windows 7

### Printer model

Sapphire Pro

lukasmatena, 2022-04-25:
Duplicate of #7036 and others. Closing.

closed 2022-04-25 by lukasmatena

ahmetcemturan, 2022-04-25:
Then maybe put in a warning in the popup dialog box?