prusa3d/PrusaSlicer #8241: Bed STL origins not aligned

kr15uk, 2022-04-25:
### Description of the bug

Added bed STL to the machine but origin seems to be off despite it shows ok in Fusion 360.
Please see attached screenshots.

### Project file & How to reproduce

[image]

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.4.1

### Operating system

macOS

### Printer model

Printrbot Simple Metal

neophyl, 2022-04-25:
How have you confirmed the stl origin ?  Besides visually in fusion that is.  How about attaching the stl file used as well as your project file so that there is at least something for the devs to work with.

kr15uk, 2022-04-25:
Sorry, GitHub didn't let me to add STL file nor 3mf, unsupported format.
Did a zip file instead, seems a bit dodgy that you have to zip it though to attach.

To verify if home position on STL is as on Fusion 360 I've simply opened the STL with Prusa slicer and origin was correct, see screenshot.

Ta.

[image]

[Archive.zip](https://github.com/prusa3d/PrusaSlicer/files/8557329/Archive.zip)

neophyl, 2022-04-25:
Zipping up files is normal on github (and the Prusa forum).
Ok figured it out.  Models placed as bed models are placed referenced from the centre.  I took your stl into Blender, centred it on the world zero and re-exported it with the middle of the object (in x/y) as 0,0.  Left z as it was.  After loading that into the bed shape it places correctly on the plater.
[image]

[simple_metal_bed_model v1_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/8557539/simple_metal_bed_model.v1_neo.zip)

kr15uk, 2022-04-25:
Brilliant, can confirm it works. Ta

closed 2022-04-25 by kr15uk