prusa3d/PrusaSlicer #8243: Add/change máx volumetric speed to máx linear extrusión speed

ManuelPeiro, 2022-04-26:
Hi,

I propose a slight change, there is a function that, at least for me, is very useful. To be able to limit the printing speed, according to the configuration of max volumetric speed. I propose that simplifying it a step further, would be to replace or add this limit with something like max linear extrusion speed.

In the end it is the same, but easier to find out... You put the extruder at the desired temperature, and you start extruding, first at low speed and gradually you go up, until you notice that the extruder loses steps. At that limit of linear extrusion, for example 30mm / s you reduce it a little for safety, 27 mm / s and that would be the value to configure in Super Slicer (without taking into account the height of layer, that which is ordered SS.

It's the same, but easier to find.... You just have to extrude faster and faster, it doesn't require any mathematical operation, even if it's simple

neophyl, 2022-04-26:
This is the PRUSA slicer GitHub, not the SuperSlicer one. Please post to the appropriate project.

ManuelPeiro, 2022-04-26:
Ooopppssss... Sorry

closed 2022-04-26 by rtyr