prusa3d/PrusaSlicer #8244: qhull build issue with git master on openSUSE Tumbleweed

frispete, 2022-04-26:
### Description of the bug

Build fails with:

```
[    6s] CMake Error at src/CMakeLists.txt:28 (target_link_libraries):
[    6s]   The link interface of target "qhull" contains:
[    6s] 
[    6s]     Qhull::qhullcpp
[    6s] 
[    6s]   but the target was not found.  Possible reasons include:
[    6s] 
[    6s]     * There is a typo in the target name.
[    6s]     * A find_package call is missing for an IMPORTED target.
[    6s]     * An ALIAS target is missing.
```

The relevant dependencies are available, of course.
```
BuildRequires:  pkgconfig(qhull_r)
BuildRequires:  pkgconfig(qhullcpp)
```

### Project file & How to reproduce

Just attempt to build PrusaSlicer on openSUSE Tumbleweed.

Here's a rpm [.spec](https://build.opensuse.org/package/view_file/home:frispete:3dprinting/PrusaSlicer-git/PrusaSlicer.spec?expand=1). 

The problem seems to be, that qhullcpp is built statically, which might be a packaging issue.
But the build looks sane and implies, that qhullcpp is build intentionally just static.

The content of qhullcpp-devel-static-2020.2-20.29 is:
```
drwxr-xr-x    2 root     root                        0 Apr 11 20:12 /usr/include/libqhullcpp
-rw-r--r--    1 root     root                    12862 Jul 24  2020 /usr/include/libqhullcpp/Coordinates.h
-rw-r--r--    1 root     root                     6989 Jul 24  2020 /usr/include/libqhullcpp/PointCoordinates.h
-rw-r--r--    1 root     root                     6690 May 23  2020 /usr/include/libqhullcpp/Qhull.h
-rw-r--r--    1 root     root                     2373 May 22  2020 /usr/include/libqhullcpp/QhullError.h
-rw-r--r--    1 root     root                     7623 Jun  2  2020 /usr/include/libqhullcpp/QhullFacet.h
-rw-r--r--    1 root     root                     4377 Jul 24  2020 /usr/include/libqhullcpp/QhullFacetList.h
-rw-r--r--    1 root     root                     4288 Jul 24  2020 /usr/include/libqhullcpp/QhullFacetSet.h
-rw-r--r--    1 root     root                     6983 Jul 30  2020 /usr/include/libqhullcpp/QhullHyperplane.h
-rw-r--r--    1 root     root                     8078 Jul 23  2020 /usr/include/libqhullcpp/QhullIterator.h
-rw-r--r--    1 root     root                    12622 Jul 30  2020 /usr/include/libqhullcpp/QhullLinkedList.h
-rw-r--r--    1 root     root                     7906 Jul 30  2020 /usr/include/libqhullcpp/QhullPoint.h
-rw-r--r--    1 root     root                     3063 Jul 24  2020 /usr/include/libqhullcpp/QhullPointSet.h
-rw-r--r--    1 root     root                    19006 Jul 24  2020 /usr/include/libqhullcpp/QhullPoints.h
-rw-r--r--    1 root     root                     4303 Jun  3  2020 /usr/include/libqhullcpp/QhullQh.h
-rw-r--r--    1 root     root                     5149 Jul 24  2020 /usr/include/libqhullcpp/QhullRidge.h
-rw-r--r--    1 root     root                    19375 Aug 13  2020 /usr/include/libqhullcpp/QhullSet.h
-rw-r--r--    1 root     root                      795 May 22  2020 /usr/include/libqhullcpp/QhullSets.h
-rw-r--r--    1 root     root                     1292 May 22  2020 /usr/include/libqhullcpp/QhullStat.h
-rw-r--r--    1 root     root                     5258 Jul 31  2020 /usr/include/libqhullcpp/QhullUser.h
-rw-r--r--    1 root     root                     4702 Jul 23  2020 /usr/include/libqhullcpp/QhullVertex.h
-rw-r--r--    1 root     root                     3729 Jul 24  2020 /usr/include/libqhullcpp/QhullVertexSet.h
-rw-r--r--    1 root     root                     2426 May 30  2020 /usr/include/libqhullcpp/RboxPoints.h
-rw-r--r--    1 root     root                     3368 May 29  2020 /usr/include/libqhullcpp/RoadError.h
-rw-r--r--    1 root     root                     3577 Jul 21  2020 /usr/include/libqhullcpp/RoadLogEvent.h
-rw-r--r--    1 root     root                     2201 Jul 30  2020 /usr/include/libqhullcpp/RoadTest.h
-rw-r--r--    1 root     root                     1539 May 22  2020 /usr/include/libqhullcpp/functionObjects.h
-rw-r--r--    1 root     root                  4168244 Apr 11 20:12 /usr/lib64/libqhullcpp.a
-rw-r--r--    1 root     root                      185 Apr 11 20:11 /usr/lib64/pkgconfig/qhullcpp.pc
```

I seem to be able to work around the issue with:
```
# fix qhull link with static lib issue
sed -i 's#INTERFACE Qhull::qhullcpp#INTERFACE -lqhullcpp#' src/CMakeLists.txt
```

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

git master @926ae0471

### Operating system

openSUSE Tumbleweed 2020424

### Printer model

Prusa Mini

tamasmeszaros, 2022-05-12:
I've looked around pkgs.org and to me it seems that the packages qhull_r and qhullcpp should use the same source package but they don't. 
See the pages and look for the entry 'Source Package' :
https://opensuse.pkgs.org/tumbleweed/science-x86_64/qhull_r-devel-2020.2-19.53.i586.rpm.html
https://opensuse.pkgs.org/tumbleweed/opensuse-oss-x86_64/qhullcpp-devel-static-2020.2-2.5.x86_64.rpm.html

libqhullcpp.a is installed into a lib64 subfolder while the qhull_r stuff is in lib subfolder. The CMake config script for qhull_r is then unable to find qhullcpp.a because it assumes it in a different location.

tamasmeszaros, 2022-11-16:
I've looked around and to me it seems that the packages are now fixed. @frispete could you please verify if the problem still persists in your setup?

lukasmatena, 2023-02-09:
No response, closing.

closed 2023-02-09 by lukasmatena

themanyone, 2024-11-27:
I had a similar problem because Fedora didn't set `LIBRARY_PATH=/usr/lib64`