prusa3d/PrusaSlicer #8246: Access variables or placeholders within the G-code Substitutions option:

justoeu, 2022-04-26:
**Is your feature request related to a problem? Please describe.**
The new functionality of replacing parts of a Gcode during the Slice process is amazing, however, it would be very interesting to be able to access the variables or placeholder to further enhance the replaces and make the replacement process even more dynamic

**Describe the solution you'd like**
The solution would be, in a simple way, but we can include the names of the variables directly in the finds or replace. For example

Find: 
   G1 E-**$${filament_retract_length}**  F3600
Replace
  ;G1 E-**$${filament_retract_length}**  F3600

In this case, could be include the symbol $$ to make easy to compile understand that, after this symbol the prusaSlicer must have replace to value of this variable or placeholder.

[image]

**Describe alternatives you've considered**
We currently have no alternatives for this feature.

foreachthing, 2022-04-26:
What are you trying to achieve?
As far as I understand is that G-Code substitutions are done after the Prusa placeholder have been parsed.
So, you would have to find "G1 E-[number] F3600" with regex: https://regex101.com/r/5aaFju/2.
I'm no regex-wiz so you (or someone else) can come up with something better than what I did.

justoeu, 2022-04-26:
Well, the idea is to make the search process more dynamic, since some options, as I mentioned earlier, are variable, such as the case of filament retraction that, depending on the type of brand, or profile, changes.

The use of a regex can be done in several ways, one of them is how you commented on number, or even do a search considering anything that starts or is between something, but the idea here would be to make it easier for those who know the variable and the value as it would be changeable, which would make it easier to write, test, or even find values.

Regarding REGEX I even know how it would be and not exactly what I meant here.

closed 2026-09-29 by prusaslicer-bot[bot]