prusa3d/PrusaSlicer #8247: Add STL thumbnail for windows file explorer

zhollis21, 2022-04-27:
Hello, 

I've recently upgraded from a FlashForge Adventurer 3 to a MK3S+ and have been very happy with the printer and mostly happy with PrusaSlicer.

One of the really nice features I miss about FlashPrint slicer is it would create a thumbnail for STL after opening it, which is really handy for looking through your STL files quickly and finding what you want.

Attached is a screenshot of my file explorer, below the green line is from when I was using FlashPrint slicer and above the line is when I started using PrusaSlicer. 
[image]

Please add this really handy feature as it makes it much easier when searching through your files to decide what to print.

bubnikv, 2022-04-27:
I suppose it will be sufficient if you install this Windows Explorer plugin.
https://3dprintbeginner.com/how-to-enable-stl-thumbnails-in-windows-10/?msclkid=733c8ebac62a11ecbc6fb5bc427546eb
I did not test it myself though.

zhollis21, 2022-04-27:
The license of that GitHub repository seems to indicate that Prusa could probably just bundle it in with their installer. That might be the easiest way to get it working out of the box.

bubnikv, 2022-04-27:
It is never supported by a slicer, only the slicer installer installs the Windows Explorer plugin.
We are evaluating this particular plugin, whether is stable enough to be included into our Windows installer.

TylonHH, 2023-05-01:
That makes no sense to me. I use the windows 3D builder to view the STLs. I also wanna open them with it. So the preview within the file explorer is handled with the default program of STL files. So this is not a Prusa Slicer issue.
If you use the slicer also for opening STL (do you?) then this should be implemented. But a slicer is (for me) not a STL viewer.

closed 2023-05-01 by zhollis21

LunarWerxs, 2026-10-16:
Thanks for raising this. SageThumbs 2K already shows Explorer thumbnails for STL files on Windows 10 and 11, and it draws them itself rather than relying on a slicer. It is free for personal use: https://sagethumbs.lunarwerx.com

Disclosure: I make it.