prusa3d/PrusaSlicer #8248: Only 1 extruder works

m-sundman, 2022-04-27:
### Description of the bug

PrusaSlicer crashes when trying to slice with any other extruder than the first one.

### Project file & How to reproduce

Steps to reproduce:
1. Add an extruder (in Printer Settings -> General -> Extruders).
2. On the empty build plate, choose Add Shape -> Box.
3. For this new part, choose to use Extruder 2 (or higher number, but not the default).
4. Press "Slice now".

At this point PrusaSlicer crashes without any error in the console.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.2-rc2+linux-x64-GTK3

### Operating system

Kubuntu 22.04 (x11)

### Printer model

Artillery Genius

m-sundman, 2022-04-27:
The same problem with 2.5.0-alpha0+418-et-opengl-es-linux-x64-GTK3.

lukasmatena, 2022-04-27:
Thanks. It also crashes in 2.2.0 and 2.3.1, so it is not a new issue in 2.4. I'll look into it.

m-sundman, 2022-04-27:
It also happens on version 2.4.2+linux-x64-GTK3 and both with "Single Extruder Multi Material" enabled and disabled.

m-sundman, 2022-05-09:
If this affects all people printing with multiple materials this problem should be huge! Did you have time to look into it?

m-sundman, 2022-05-10:
It's been working just fine in all versions until very recently. I haven't been printing multi materials for a couple months, but before that it was working just fine. It's very weird that it stopped working suddenly, because I haven't changed that custom gcode sections since it was working. I think. (Or there's a chance I've copied something over from SuperSlicer to Prusa Slicer and then somehow not testing it in Prusa Slicer.) How's that supposed to look if I want to get the name of the next extruder?

Btw, it also crashes if those custom gcode lines are outcommented, which is weird, because the code really shouldn't go touch comments at all.

Anyway, super big thanks for providing the workaround of just removing that custom gcode, or replacing them with klipper macros or something!

lukasmatena, 2022-05-11:
@m-sundman
"In all versions until recently" is not very helpful. Giving an exact version where you tried and succeeded would be.

>  the code really shouldn't go touch comments at all

Why not? Macro expansion works inside comments too, you are using it yourself: `; L [layer_num]`.

> How's that supposed to look if I want to get the name of the next extruder?

It's not supposed to work. PrusaSlicer never allowed to name an extruder, the variable does not exist. It really seems that the Custom G-Code is in fact from SuperSlicer.

The only issue I see is that the application crashes instead of reporting an error.

m-sundman, 2022-05-13:
@lukasmatena Yes, you're right on all accounts. (Btw, I tried to find what variables are available in prusa, but in the end just found a small subset, and then random further details in some random forum posts. It would be nice with a complete list of existing variables and description of how they work.) And yes, I'd consider this ticket fixed if PrusaSlicer would show a proper error message stating the problem (accurately enough for the user to find what configuration to change to fix it) rather than crashing.

lukasmatena, 2022-05-13:
> would be nice with a complete list of existing variables and description of how they work.)

You might find  https://help.prusa3d.com/article/macros_1775 helpful.

m-sundman, 2022-05-15:
> > would be nice with a complete list of existing variables and description of how they work.)
> 
> You might find https://help.prusa3d.com/article/macros_1775 helpful.

That has only some examples. And https://help.prusa3d.com/article/list-of-placeholders_205643 apparently also has only some examples. It would be very nice to have an automatically generated (and thus up-to-date) list of variables.

lukasmatena, 2022-05-17:
@m-sundman Can you please give an example of something that is missing in that documentation? Thanks.

m-sundman, 2022-07-08:
@lukasmatena I think the "missing" values were values from the settings, but since they can be seen in the program itself I suppose they're not missing as such.

lukasmatena, 2024-02-19:
Closing, I think it has been resolved.

closed 2024-02-19 by lukasmatena