prusa3d/PrusaSlicer #8249: PrusaSlicer locks up during slicing

AttilaSVK, 2022-04-27:
### Description of the bug

When trying to slice two specific files (which do work fine in Cura 4.13.1), PrusaSlicer hangs at 20% (generating perimeters) while using up all the CPU resources.

### Project file & How to reproduce

1) Load up this specific STL (or there is a smaller version for smaller printers, which I tried and included as well)
2) Select any profile and material
3) Hit the Slice button

If I cut the object under the spiral part and remove it, then it slices fine, but obviously the object is useless that way...

[gnal_100ft_spiral.zip](https://github.com/prusa3d/PrusaSlicer/files/8573086/gnal_100ft_spiral.zip)
[gnal_50ft_spiral.zip](https://github.com/prusa3d/PrusaSlicer/files/8573109/gnal_50ft_spiral.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.4.2+win64 / 2.4.1+arm64

### Operating system

Windows 10 Pro 21H2 / macOS 12.3.1

### Printer model

Prusa i3 MK3S / Custom built printer based on Alfawise U20 frame

neophyl, 2022-04-27:
Sounds like a duplicate of this one https://github.com/prusa3d/PrusaSlicer/issues/8164

AttilaSVK, 2022-04-27:
Tried turning off the Fill gaps option, the issue is still present (at least on 2.4.2+arm64 on macOS 12.3.1)

black-fury, 2022-04-28:
I have the same issue with [Smallest Wallet - Coin Purse](https://www.thingiverse.com/thing:3216321/files) file on Slicer version 2.4.2+win64, same Windows 10 version, Printer: Prusa Mini+. Tried turning off the Fill gaps option, but the problem still remains. No issue with other models though.

kubispe1, 2022-12-17:
Hi, in the upcoming 2.6.0 there will be improvements in the performance of the Medial Axis algorithm. The calculation should be much more efficient. 
For now, I am closing this issue. Thank and have a nice day.

closed 2022-12-17 by kubispe1