prusa3d/PrusaSlicer #8252: GCode Viewer Crashes with Fuzzed Input

powpingdone, 2022-04-28:
### Description of the bug

I was doing fuzzing for a assignment for one of my classes, and I decided to do some basic fuzzing of PrusaSlicer's GCode Viewer. I found 3 gcode files that crashed that logged with a similar message:
```
--------------------------------------------------------------------------
Message: 20:11:42: Starting PrusaSlicer flatpak with entrypoint script
--------------------------------------------------------------------------
[2022-04-27 20:11:42.298308] [0x00007f2db9a679c0] [trace]   Initializing StaticPrintConfigs
Gtk-Message: 20:11:42.326: Failed to load module "canberra-gtk-module"
Gtk-Message: 20:11:42.326: Failed to load module "canberra-gtk-module"
[2022-04-27 20:11:43.705691] [0x00007f2db9a679c0] [error]   Unable to create bed triangles
/usr/include/c++/11.2.0/bits/stl_vector.h:1045: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = Slic3r::Polygon; _Alloc = std::allocator<Slic3r::Polygon>; std::vector<_Tp, _Alloc>::reference = Slic3r::Polygon&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__n < this->size()' failed.
```

### Project file & How to reproduce

[for_pslicer.zip](https://github.com/prusa3d/PrusaSlicer/files/8578741/for_pslicer.zip)

cube.gcode is a known good gcode file, the rest are ones that crash.

To Reproduce: 
1. Open any of the bad_*.gcode files.
2. Let PrusaSlicer GCode Viewer crash.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.4.1 from flathub

### Operating system

Fedora Silverblue 36.20220422

### Printer model

Prusa MK3S

jmd73, 2022-04-28:
@powpingdone Hi!
The bad files are really bad files. Somthing seems to have messed up when saving the files or something. For example the header in the working cube file is like this:
; generated by PrusaSlicer 2.4.1 on 2022-04-23 at 01:29:52 UTC

; 

; external perimeters extrusion width = 0.45mm
; perimeters extrusion width = 0.45mm
; infill extrusion width = 0.45mm
; solid infill extrusion width = 0.45mm
; top infill extrusion width = 0.40mm
; first layer extrusion width = 0.42mm

While the header in the broken cube file is like this:
;££ generated y
PrusaSliceÉÉÉr 2.4.1 on 2022-04-2
:at 01:29:52 UTC

; 

; Oxternal peri***meters extrusion width g 0.45m
; perimters extrusio wi¢h = 0.45mm
; infill eUUUxtrusion width = 0.45mm
; olid infill extrusion width =¬¬¬¬ 0.45mm
; top infill extrusion widh = 0.40mm
; first layer etrusion width = 342mm

The other files are missing the header part completely and the g-code that is in the files are mixed ascii and non ascii characters. The ascii parts seems to have really bad values too.

As an example this is from the beginning of the file bad_generated.gcode:
M201 E3320561040.632314 T-519283557.1735166 X4067116008.5240545 Y905153298.0535586 Z2745693542.4162374 
G28 
M82 
M115 
M109 B2056677510.6973917 F1656168090.4833963 I1005489973.4060211 R1609625286.8637679 S-67357446.3376701 T3506033570.223804 
M205 B2868827536.6088805 E-1619512010.9324503 J1804525534.8903587 S-119215853.96242183 T-2552583800.736939 X3332241824.701853 Y2205242085.1476383 Z1934830630.1689007 
M205 B-1382299165.8423307 E2715478857.6433973 J-1975809228.7586777 S1197728515.9678736 T-3605485171.7704434 X-2605116285.7030087 Y-2441421891.8096075 Z3769750280.845207 
G91

As you can see, the files are corrupted and the g-code viewer can't handle them.

powpingdone, 2022-04-28:
Of course. They were intentionally corrupted, that was part of the test.

powpingdone, 2022-11-04:
Closing this, as it is bad practice to do these sorts of drive by issues.

kubispe1, 2022-12-17:
Hi, thank you for your report. I don't get your conversation... :D :D 
But I want to inform you:  
We add a fix to your issue to the upcoming 2.6.0. when importing invalid or corrupted G-code is imported, a crash of the G-Code viewer should not appear. A new warning window announces the imported file issue.

closed 2022-12-17 by kubispe1